Mr-Markus/ZigbeeNet #5: Setup CI to avoid uncompileable code

eiaro, 2018-10-17:
Setup a CI to avoid code that cannot compile to enter master branch.

eiaro, 2018-10-17:
I am not able to setup this for you. If you choose to use AppVeyor you just access their page and activate it using your GitHub credentials. Then on the Projects page you add this repository.

Then add a file named `appveyor.yml` with the following content and it should be setup to build.

```
version: 1.0.{build}
image:
  - Visual Studio 2017
  - Ubuntu1804
install:
  - git submodule update --init --recursive
before_build:
  - nuget restore unpi-net/src/UnpiNet.sln
  - nuget restore src/ZigbeeNet.sln
build:
  verbosity: minimal
  project: src/ZigbeeNet.sln
```

Mr-Markus, 2018-10-17:
ok, i created an appvayor account and linked it with my repository. After that i created the appvayor.yml file in the repository and received that:

Build started
2
git clone -q --branch=master https://github.com/Mr-Markus/ZigbeeNet.git C:\projects\zigbeenet
3
git checkout -qf a14f1fcc1aca548079554bb299c59a4f3d581cf8
4
Solution project file "UnpiNet.csproj" cannot be found.

Was everything done correctly and could you use it or do i have to do further steps?

eiaro, 2018-10-17:
I can correct the build file and take it from here.

Mr-Markus, 2018-10-17:
I replaced unpi-net Project reference which came from submodule with new NuGet Package

After the commit it build automatically with Status succeded, but i think it dind't worked:

[00:00:00] Build started
[00:00:00] git clone -q --branch=master https://github.com/Mr-Markus/ZigbeeNet.git C:\projects\zigbeenet
[00:00:05] git checkout -qf 395e7fd19d80ee5d0831e9abadc28a8260e56a58
[00:00:06] msbuild "C:\projects\zigbeenet\src\ZigbeeNet.sln" /verbosity:minimal /logger:"C:\Program Files\AppVeyor\BuildAgent\Appveyor.MSBuildLogger.dll"
[00:00:08] Microsoft (R) Build Engine version 14.0.25420.1
[00:00:08] Copyright (C) Microsoft Corporation. All rights reserved.
[00:00:08] 
[00:00:12] C:\projects\zigbeenet\src\ZigbeeNet\ZigbeeNet.csproj.metaproj : warning MSB4078: The project file "ZigbeeNet\ZigbeeNet.csproj" is not supported by MSBuild and cannot be built.
[00:00:12] C:\projects\zigbeenet\src\ZigbeeNet.ZCL\ZigbeeNet.ZCL.csproj.metaproj : warning MSB4078: The project file "ZigbeeNet.ZCL\ZigbeeNet.ZCL.csproj" is not supported by MSBuild and cannot be built.
[00:00:12] C:\projects\zigbeenet\src\ZigbeeNet.CCZnp\ZigbeeNet.CC.csproj.metaproj : warning MSB4078: The project file "ZigbeeNet.CCZnp\ZigbeeNet.CC.csproj" is not supported by MSBuild and cannot be built.
[00:00:12] Discovering tests...OK
[00:00:12] Build success

Could you check it pls? Thanks

Mr-Markus, 2018-10-18:
Fixed by #8

closed 2018-10-18 by Mr-Markus