Mr-Markus/ZigbeeNet #6: Remove referenced projects that isn't in the project

eiaro, 2018-10-17:
Reference to local projects will make project hard to maintain. Either include it in the project, or separate it out as a own project and publish a Nuget for it.

eiaro, 2018-10-17:
You probably wouldn't use this project unless you use unpi-net or another hardware interface, so in my mind they belong together. If another hardware interface is needed, it should be added accordingly.

A end user should probably only need to do:
```
var zigbeeInt = ZigbeeNet.CC2531(); // Static function to initialize the interface
zigbeeInt.PacketReceived += MyPacketHandler;
zigbeeint.Send(amazing stuff);
```

If the UNPI is intended to be a separate project for other implementations, then it should be moved out as it's own project when the need arise.

A cool feature is to include the possibility of using the interface over a TCP connection (serial to TCP) in addition to regular serial port.

eiaro, 2018-10-17:
Yes, the UNPI can be used for other projects, but currently it is only used here. So pragmatically you should include it in the project, not make a separate project (repository) for it.

At a later stage you may take it out of this project and make it a separate project, but when you do, it should be testable that it doesn't break any implementations (this project). It's a common problem when working on a project that is divided into parts; when do you split them up. Answer is usually _when needed_.

In the end, by having it in the project, if only temporary, will make development easier and refactoring has less chance of breaking.

Mr-Markus, 2018-10-17:
Yes, you are right. Tipically Problems you have during new development.

The unpi-net library is now at a Version that is running and don't need further development to run ZigbeeNet. We are now at the point where logic needs to be implemented inside ZigbeeNet, because sending data to the TI Hardware via unpi-net is running.

So i think it that the next step would be to create a nuget package for unpi-net and reference it in ZigbeeNet.

Mr-Markus, 2018-10-17:
I removed the submodule and replaced project reference of unpi-net with it's new NuGet package.

closed 2018-10-17 by Mr-Markus