Mr-Markus/ZigbeeNet #10: Logger Interface

Mr-Markus, 2018-10-19:
For better logging experience it would be great if the caller of ZigbeeNet could choose his own logger library.

So an interface with common Methods like Log(LogLevel level) could be a possible solution.

Ideas are wellcome at all time :-)

nicolaiw, 2018-10-19:
I think that is not necessary. Serilog already has many sinks and is therefore very flexible. I think we should not reinvent the wheel here. However, I would offer the caller the possibility to pass a logger configuration which is currently hardcoded in CCZnp.Init ().

eiaro, 2018-10-19:
LibLog is made for the purpose.

nicolaiw, 2018-10-19:
I just had a look at LibLog and it seems to be exactly what we are looking for.

@Mr-Markus do you agree ?

nicolaiw, 2018-10-19:
```
using System;
using Serilog;

namespace` ZigbeeNet.PlayGround
{
    class Program
    {
        static void Main(string[] args)
        {
            // Configure Serilog
            Log.Logger = new LoggerConfiguration()
               .MinimumLevel.Debug()
               .WriteTo.Console()
               .CreateLogger();

            try
            {
                var zigbeeService = new ZigbeeService(new Options { Baudrate = 115200, Port = "COM3" });
                zigbeeService.Start();
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.ToString());
            }

            Console.ReadLine();
        }
    }
}
```
will now produce something like:

[image]

I removed all dependencies of Serilog and instead added LibLog. (@eiaro thanks for your advice). Now the user of the lib can decide which framework to use or even write a custom logger.

@Mr-Markus something happend that should not :) .. I unintentionally pulled directly to the project instead of sending a pull request. Sry for that! To avoid this in the future pls remove me as a contributor so that I can't pull directly to this repo.

Mr-Markus, 2018-10-19:
Yes, exactly what we were looking for!

Thanks for Suggestion [@eiaro](https://github.com/eiaro) and thanks to [@nicolaiw](https://github.com/nicolaiw) for implementation

[@nicolaiw](https://github.com/nicolaiw)
Ok, i would remove you from the contributor list and you can create pullrequests for your changes :)

Because of that i will close the issue. Thanks!

closed 2018-10-19 by Mr-Markus