Mr-Markus/ZigbeeNet #12: Add docs, release notes and proper versioning

nicolaiw, 2018-10-19:


Mr-Markus, 2018-10-19:
I updated the README.md file with more details

nicolaiw, 2018-10-20:
Looks nice. We should also consider adding a list of the supported Zigbee features (nodes, commands, clusters,..). I would suggest to not close this issue until we get a first stable version.

Just a little question: are you allowed to use the Zigbee logo in the readme? (copyright)

nicolaiw, 2018-10-20:
Sounds nice.

Mr-Markus, 2018-10-21:
You can download the Z-Stack including Software/Tool/Document for free on TI's Website at [http://www.ti.com/tool/z-stack](http://www.ti.com/tool/z-stack )

But you Need to register before you can download it. After that i received following:

> I certify that the following is true:
> (a) I understand that this Software/Tool/Document is subject to export controls under the U.S. Commerce Department’s Export Administration Regulations ("EAR").
> 
> (b) I am NOT located in Cuba, Iran, North Korea, Sudan or Syria. I understand these are prohibited destination countries under the EAR or U.S. sanctions regulations.
> 
> (c) I am NOT listed on the Commerce Department’s Denied Persons List, the Commerce Department's Entity List, the Commerce Department’s General Order No. 3 (in Supp. 1 t o EAR Part 736), or the Treasury Department’s Lists of Specially Designated Nationals.
> 
> (d) I WILL NOT EXPORT, re-EXPORT or TRANSFER this Software/Tool/Document to any prohibited destination, entity, or individual without the necessary export license(s) or authorization(s) from the U.S. Government.
> 
> (e) I will NOT USE or TRANSFER this Software/Tool/Document for use in any sensitive NUCLEAR, CHEMICAL or BIOLOGICAL WEAPONS, or MISSILE TECHNOLOGY end-uses unless authorized by the U.S. Government by regulation or specific license.
> 
> (f) I understand that countries other than the United States may restrict the import, use, or export of the Subject Product. I agree that we shall be solely responsible for compliance with any such import, use, or export restrictions.
> 
> I / We hereby certify that we will adhere to the conditions above.
> I / We do not know of any additional facts different from the above.
> I / We take responsibility to comply with these terms. 
> I / We understand we are responsible to abide by the most current. versions of the Export Administration Regulations and other U.S. export and sanctions laws.

And because of (d) i think we should remove the TI documents from the repository and just reference public Z-Stack site

Mr-Markus, 2019-02-06:
Files added into here [https://github.com/Mr-Markus/ZigbeeNet/tree/master/docs/zigbee-protocol](https://github.com/Mr-Markus/ZigbeeNet/tree/master/docs/zigbee-protocol)

closed 2019-02-06 by Mr-Markus