Mr-Markus/ZigbeeNet #9: Abstract Hardware Interface

nicolaiw, 2018-10-19:
A end user should probably only need to do:
```
var zigbeeInt = ZigbeeNet.CC2531(); // Static function to initialize the interface
zigbeeInt.PacketReceived += MyPacketHandler;
zigbeeint.Send(amazing stuff);
```

If the UNPI is intended to be a separate project for other implementations, then it should be moved out as it's own project when the need arise.

A cool feature is to include the possibility of using the interface over a TCP connection (serial to TCP) in addition to regular serial port.

_Originally posted by @eiaro in https://github.com/Mr-Markus/ZigbeeNet/issues/6#issuecomment-430584007_

nicolaiw, 2018-10-24:
I currently do not have any Hardware for testing but already ordered one. May be we could have a look at https://github.com/zsmartsystems/com.zsmartsystems.zigbee (java). This project also includes code for hardware abstraction. One of the supportet hardware is the cc2531 (https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/master/com.zsmartsystems.zigbee.dongle.cc2531). It looks well structured and may be we can take over some of the Ideas.

Mr-Markus, 2019-01-24:
done!

closed 2019-01-24 by Mr-Markus