Mr-Markus/ZigbeeNet #16: Code style

eiaro, 2018-10-21:
If there is a preferred code style, please set up a ReSharper rules file for it and I will follow.

Mr-Markus, 2018-10-21:
I also don't have ReSharper. But i can agree @nicolaiw sugesstions.

One thing i would prefer, because of better readability:
- use data type on variable declaration instead of var

```
int a = 1;
int b = 1;
```

so use:
`int c = a + b;`
instead of
`var c = a + b;`

eiaro, 2018-10-21:
The var statement was introduced in C# 3.0 to improve readability...

I don't argue format, so whatever you decide.

Mr-Markus, 2018-10-21:
Most time we are reading code. So readability is a fundamental point.

var is good when you are working with Linq for example, but for declaration of variables does not cost more time to write data type, but improves readability

I won't get in discussions but it would be a help if you use implicit type of it is not clear on declaration. 

Otherwise it will be OK if you write things like var count = 0 where it is clear which datatype it is

nicolaiw, 2018-10-21:
@eiaro Im with you.. MS introduced it to improve readability..but in this case I guess we can benifit of explicte type declarations.. Zigbee is not an easy protocol and when someone tries to contribute it could be helpfull to see the type at the first sight without hover the method to see which type u are get from a method or what else.. Zigbee is not easy so I for my self would also prefer to declare types explicitly instead of using var

eiaro, 2018-10-21:
For what it's worth, I disagree. In 10 years on the Net platform, I've never heard anyone wishing back to explicit declaration. 

If you want contributors to follow your guidelines you must create some sort of style sheet. ReSharper is good, but there is also https://github.com/DotNetAnalyzers/StyleCopAnalyzers

Mr-Markus, 2018-10-21:
Ok following… because we do not have ReSharper and other Tools at the Moment we can not provide some guidelines at the moment.
Your code isn't bad and it was just a suggestion. 

This should not blocks our development, because we have a great software we are working on and i am happy to work with you :-)

This issue is now on-hold

eiaro, 2018-10-21:
You don't have this in your VS?
[image]

Write the settings in a file and name it `.editorconfig` and it will be applicable for everyone on the project if they work on an editor supporting EditorSettings (most do).

commit 5947fa72 (2019-02-06): coding-guidlines added. closes #16
closed 2019-02-06 by Mr-Markus