Mr-Markus/ZigbeeNet #18: Process AREQ SerialPacket as Response

Mr-Markus, 2018-10-21:
https://github.com/Mr-Markus/ZigbeeNet/blob/0d06cf9d5b402de25a95c8ea38ce89e1ff84090e/ZigbeeNet/CC/CCZnp.cs#L403

At this Point it can happen that a packet with type AREQ can be read from InputStream. This is defined by TI in UNPI

Asynchronous Request transfer initiated by Host Asynchronous Indication – transfer initiated by NP. Both types of asynchronous messages use the same message type bit field (0x02). The type of message (request/indication) depends on which processor initiated the transaction.

Two Scenarios:
1. You send an SREQ and will return a SRSP with Status. But as second Response you receive an AREQ packet, too.
Example:  nodeDescReq which is send as SREQ but awaits a SRSP with Status if it was sent correct and a second AREQ that contains device info
This is an AREQ packet that is send from client-to-server  (Server = end device)

2. An end device in yout Network (e.g. Thermostat) is sending you new Information because Temperatur changed. 
This is also an AREQ packet. But is is send from server-to-client  (Server = end device)

For Scenario 1 i already created a JSON mapping file [zdo_ReqRspMap.json](https://github.com/Mr-Markus/ZigbeeNet/blob/master/ZigbeeNet/CC/Defs/zdo_ReqRspMap.json) that defines which SREQ awaits which AREQ packet, too. In class file ZpiObject the property `public ZpiObject IndObject` holds the Information if there is an AREQ Response awaited

I think this Information  have to be used here

Mr-Markus, 2018-10-21:
@eiaro 
Are you working at this point? I would like to avoid conflicts because of working at the same Code at the same time.
But It is fundamental for further work. If so I will try T4 template with @nicolaiw within the next days

eiaro, 2018-10-21:
At the moment not working. I have no clue on how to read the files, and their names doesn't match the spec document so I am at a loss. See #19 for status.

Mr-Markus, 2018-10-21:
Yeah names are different, becaue i got this file from another project and i don't know why they changed it, but because of the huge amount of commands etc. i haven't changed them and search by id or parts of Name by it...
That is the reason of the Meta classes and the generic ZpiObject. But I will work on them and also try if it is possible to enhance that files or convert it to another format.
Locally I already have created some toolt that do parsing it. With this Tools i created the Meta classes. It should be possible to convert them.

closed 2018-10-24 by Mr-Markus