Mr-Markus/ZigbeeNet #29: Binding a Zigbee device

thombrink, 2019-01-21:
Hello,

I saw your code last week and have implemented the level/brightness cluster.
(pull request is coming when the remaining problems are solved)

The 'only' problem is that I can't seem to get it working.
I believe I have to bind my device (Ikea Tradfri lamp), before it accepts the commands I send to it.

With debug logging enabled, I can see incoming DeviceAnnounce messages. But I think its not enough to just process this message and let the device be added to my nodes.

In the cluster class I also found a Bind method, but this also does not seem to work. 

Is there a possibility you could help me with this?

Mr-Markus, 2019-01-24:
Hello,

if there is a new device announced the library creates an instance of the NodeDiscoverer class to retrieve necessary information of the device like the NetworkAdress, implemented clusters etc. and adds the node to an internal list.
With these information it should be possible to send messages to the node without calling a BIND_REQUEST. As I understand the Bind Request is used to send messages without knowing the NetworkAddress. (Please correct me if it is wrong)

So you should be able to use your implemented Cluster like this:

```
networkManager.AddSupportedCluster(0x06);

var node = networkManager.GetNode(addr);
if (node != null)
{
        ZigBeeEndpoint ep = new ZigBeeEndpoint(node, 0);
        node.AddEndpoint(ep);

        ZclOnOffCluster onOff = new ZclOnOffCluster(node.GetEndpoint(0));

        onOff.ToggleCommand();
}
```

That's the way it should work, but I still have problems with getting information on DeviceAnnounce Message with CC2531 😕

Maybe the Zigbee Link Key is not implemented correctly with this Hardware. As you can see in the BasicSample Project the LinkKey is set in the demo Program. Do you have deeper understanding in it?

If you have any idea I would be very happy about your PullRequest

Mr-Markus, 2019-01-26:
Ok but it makes sense. I will take a look on it.

As you said it should not be the problem and there is another point of code which is relevant. 

Do you have already Zigbee Hardware?

The architecture of this library is designed to use it with different gateways. CC2531 is one on many and after i began to start with it I realized that it is the most complexe. I am at the point that I would start to implement another gateway like XBee, because some steps like pairing make me going crazy...

But I do not give up and will further work on this project.

Mr-Markus, 2019-01-29:
Thanks for your pullrequest. I merged it into the master branch.

I tested the PlayGround example with an Ikea Tradfri lamp and it works!
I toggled the lamp a few times by switch so that it was in the pairing mode, then the device was announced in the console and displayed the network address and with that address i was able to execute the implemented toggle, level and move commands.

I also tried to disconnect the lamp from the mains and connect it again and it worked again and the network address was still the same.

Does it also work with your hardware?

thombrink, 2019-01-29:
I'm sorry, I forgot to reply.

I have a CC2531 gateway, and I'm thinking about testing with a CC2530 too.
You can find a lot of info about pairing etc on this repo: [https://github.com/Koenkk/zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt).

He has a great wiki for the CC2530 and CC2531 (including [range comparison](https://github.com/Koenkk/zigbee2mqtt/issues/52))

The playground was my test project for some commands as you can see. The toggle,level and move commands both work fine with my hardware (also a Ikea Tradfri).

To get everything working I started the playground, switched the light off and on to register it and typed a command in the console window.

Before requesting the pull request I removed the code that automatically added the node by address, so I didn't had to keep turning the light off and on while testing.

What is your next objective (with this repo)?

Mr-Markus, 2019-01-30:
Thanks for this informations. It will help a lot for a better understanding.

As next steps I think it is the best to implement other clusters, beginning with the Generals and Ligthing 
The clusters are described in the official Zigbee document (Chapter 3 and 5):
[https://github.com/Mr-Markus/ZigbeeNet/blob/master/docs/Zigbee-cluster-library-specification.pdf](https://github.com/Mr-Markus/ZigbeeNet/blob/master/docs/Zigbee-cluster-library-specification.pdf)

If there are some clusters it would be great if another hardware like XBee will be implemented.

The target of this repo should be to provide a library for working with Zigbee with .netstandard or .net core on different plattforms. I do not plan to develop an application or smart home system here.

It would be interesting to know what do you plan to do with this library. Maybe we have some common topics.
I also would be pleased if you help me to implement some clusters and do some further work on this repo 😊

thombrink, 2019-02-01:
I don't like the idea of relying om software companies or open source projects to control my home. Therefore I would like to build my own sort of smart home system.

It's not realistic to expect that this is something that's done in a few hours, resulting in the same features and stability.

My idea is to create a google assistant (actions) project to control my home first, so I can work out my own system in the meantime. I would like to create a nuget package from this repo to use in my google assistant/smart home project.

Because I only have one Zigbee device at the moment, I can't really test the cluster feature. As soon as I have another device (could take a while), I will make it work. If you're impatient I'm happy to assist you if there are any questions about implementing the clusters.

Mr-Markus, 2019-02-05:
All right, I understand. I would create an issue for creating a Nuget package if there is the first running version with node discovery and would mention you in it.

This issue here should now be closed, right

closed 2019-02-05 by Mr-Markus