Mr-Markus/ZigbeeNet #33: Create NuGet Packages

Mr-Markus, 2019-02-06:
There should be NuGet packages for the ZigBeeNet libraries on nuget.org

They should be provided with lowest possible version

See #32 which have to be done first

Following packages should be created first:
- ZigBeeNet as netstandard2.0
- ZigBeeNet.Hardware.TI.CC2531

Gallimathias, 2019-06-03:
I find the following points should be noted and discussed:

* Auto create only on the master branch. If you want to have nightly builds you could still develop as pre-releases builds.
* If we want to have a large nuget package or separated nuget packages.

nicolaiw, 2019-06-04:
**According to point one:**

I would agree.

**According to point two:**

I would suggest to deliver multiple packages. As we plan to support more dongles/hardware in the future. Otherwise the delivered package might unnecessarily big as we would deliver packages which the consumer  may not need.

Best regards.

Mr-Markus, 2019-06-04:
I agree, too. Only master branch should be created as NuGet package at first.

I would also suggest to create multiple packages so that the caller can focus on used dongle.

jgmdavies, 2019-06-04:
@Mr-Markus 
I'm not sure about the latter point - some folk will have more than one dongle, or maybe are developing apps that are dongle-agnostic?
Best,
Jim

Mr-Markus, 2019-06-04:
@jgmdavies 
Sure, that should be possible for the user. In that case the user can reference more than one dongle package

This should not be a problem, isn't it?

jgmdavies, 2019-06-04:
@Mr-Markus 
Thanks Markus.  Well, I agree with your earlier point that in general terms the main package shouldn't be unnecessarily big.  But multiple packages would add to the maintenance work/angst - I think I'd favour the simplicity of a single package for now unless the dongle type count increases much.
I think it would be great to get any NuGet package(s), so I don't want to distract you - thanks to all for the excellent work!
Jim

Mr-Markus, 2019-06-04:
@jgmdavies 
Yeah, I agree that multiple packages will increase maintenance.

Maybe it will be a solution to create a complete package like "ZigBeeNet.All" package and seperate packages for each library, too.
This could be helpful for embedded systems, where storage is very strong limited

Thanks for compliment :-)

jgmdavies, 2019-06-04:
;---)

Mr-Markus, 2019-06-24:
I am thinking about using gitversion. Already tried it here: https://github.com/zigbeenet/ZigbeeNet/tree/gitversion

Mr-Markus, 2019-08-20:
Done. See here https://www.nuget.org/profiles/ZigbeeNet

closed 2019-08-20 by Mr-Markus