Mr-Markus/ZigbeeNet #41: Digi Xbee support

andreasfedermann, 2019-03-08:
Hi there,

do you plan to support the digi Xbee device too? Looking at #29 you will. But is there a timeline when you will begin implementing it? 
Really great work so far guys!

Let me know if you need help.

Mr-Markus, 2019-03-08:
Hi Andreas,

yes there are plans to implement the XBee hardware as next step, too. But we do not have a timeline for this, yet.
It should easily possible to do this, because the library is already designed for hardware seperation.

At the moment we don't have any hardware, yet so this would be the first step. We thought about buying the XStick, but until now we are not sure which hardware specific differences are between the XBee devices and what it means for the implementation.

Do you already have one and can tell us something about them?

Thanks

andreasfedermann, 2019-03-08:
I own an S2C device with the XB24C product family and form factor TH (Through Hole). I use it in conjunction with a waveshare arpi600 board which extends my Raspberry PI 3B+. I don't know what the exact differences between the XStick and Xbee devices are either. I tested the Xbee device with the X-CTU from digi to control my philips hue. That already worked. But because of the lack of existing .net libraries I paused my project some months ago.
But now its worth it to reactivate my project and test your lib. Maybe I could try to implement the Xbee hardware if you agree with it.

Mr-Markus, 2019-03-09:
Ok, I understand.

I would appreciate if you would try to implement it and would help you at any time if you need help.

As I told before the libary is still designed for further hardware implementation. Your main entry point would be a Manager or dongle class that implements the `IZigBeeTransportTransmit `interface.

As an example you can use the [ZigBeeDongleTiCc2531](https://github.com/zigbeenet/ZigbeeNet/blob/master/src/ZigBeeNet.Hardware.TI.CC2531/ZigBeeDongleTiCc2531.cs) class.
It would also be helpful for you to look here:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/master/com.zsmartsystems.zigbee.dongle.xbee
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tree/master/com.zsmartsystems.zigbee.dongle.xbee.autocode

The architecture at this repo is very well and the abstraction is implemented very well too, so that we created this repo with .netstandard technology and translated it mostly from java to c#
So it should be much easier for you to understand it. Otherwise don't hesitate in contacting me 😊

Mr-Markus, 2019-03-09:
Here the project structure I would suggest for your implementation:

[image]

We use Shared Projects for the implementation in the src folder and create a specific target framework project in targets folder which references to it, so that we can support muliptle plattfroms

andreasfedermann, 2019-03-09:
I really like that shared project approach. I only used linked files until now. But I have one question. What about other product families which are not a dongle. Shall these products be supported in the future too? Maybe they are already supported with the reference (java) implementation. I am struggling a bit with the naming of the class. The word dongle in the class name is maybe not sufficient enough. Shouldn't the name be more neutral? Maybe ZigBeeDigiXBee?
I want to provide unit tests as well. Do you prefer any unit test framework?

Mr-Markus, 2019-03-10:
Yes, I would like to get this library as flexible as possible. I already saw IP Gateways from Digi for example and it would be great if they would also be supported. Did you mean that?

I am not really sure about Xbee plattform and it's protocol but if the only difference would be the transmit then the "Dongle" class should be named more natural.

We already talked about Unit tests and prefered XUnit in issue #20 but haven't realised them until now :-)

Mr-Markus, 2019-04-03:
@andreasfedermann 
How is the current status for the implementation?

Please see https://github.com/zigbeenet/ZigbeeNet/issues/44 for important changes to the project structure

andreasfedermann, 2019-04-03:
Hi Markus,

I’m currently working on the auto code project which is based on system codedom. I created a fork for that. When do you think is it appropriate to create a pull request? Should I finish the complete implementation including the digi xbee stack?

I’ll merge the latest changes into my fork as soon as possible.

Mr-Markus, 2019-04-03:
Sounds great!

I think that it would be the best way to create a PR if all work is done. Otherwise we have not-runnable code in the master branch

**A short tip for the change from shared project to netstandard:**
I created a new netstandard library in another directory with same name as the shared library project, copied only the new *.csproj into the same directory as the *.shproj file and replaced the reference via editor in the sln file with the new *.csproj file and removed the old *.shproj files from the project.
And that's it 😊

nicolaiw, 2019-04-09:
@andreasfedermann  
First at all. Thank you for your effort. I just had a look at your work and just in case you didn't noticed...could you please shift your ZigBeeNet.Digi.XBee.CodeGenerator project to the src folder before sending a PR ?

Again.. Thank you!

Mr-Markus, 2019-05-01:
Hi @andreasfedermann ,
I took a look on your fork and it's code and it looks great. How much work do you think is still needed for a PR?

One thing: The path **ZigbeeNet/targets/ZigBeeNet.Hardware.Digi.XBee.core2.2/** and it's content looks like remains from the old shared project structure. Please remove it before you create a PR

Many thanks!!!

andreasfedermann, 2019-05-01:
Hi @Mr-Markus,
I’m almost done. I extended the playground project to be able to choose the dongle by arguments. I will commit and sync this soon. This week for sure. I already tested the digi xbee device but I still have some problems. If I start my coordinator and then switch on a router device (hue bulb) I get notified. My nodes get updated also. But if I try to switch off the light it is not working yet because of missing endpoint information. I think it is an configuration thing but I’m not sure. So I think the PR will be soon.
I will remove the old structure immediately too.

Best regards!

andreasfedermann, 2019-05-03:
Hello @Mr-Markus,

I tested my device and it works now. I found out that the reason it didn't work was the fact that the deserialization method of the ZdoCommand was commented out. The uncommenting was done with commit 5c401b2d. If I commented this line in again, it worked for me. Can you please tell me why the uncommenting was done? Nevertheless i will create a PR.

Mr-Markus, 2019-05-05:
Hi @andreasfedermann 
the uncommenting was done because of a bug in the CC2531 implementation. It is fixed in commit 481a6a5a5a779f3e5bb17fdee268e6bb1d3fe3c1. Thanks for your tip.
I will check the PR and will merge or comment it if there are any questions

nicolaiw, 2019-05-05:
@andreasfedermann 
Thank you very much for your work! Could you tell us which devices and commands you successfully tested with the Digi Xbee Hardware so we can adjust the readme and wiki pls?

Thank you

Best Regards

andreasfedermann, 2019-05-07:
Hey there,
the port of the code was really fun. And I will participate in the future too. So please let me know if help is needed. 
I will leave a comment regarding the tested device on the foreseen issue.
I will close this issue if this is ok for you.

nicolaiw, 2019-05-07:
I think it's ok to close this issue.

Thank you!

closed 2019-05-07 by andreasfedermann