Mr-Markus/ZigbeeNet #46: Avoid while (true) { .. } during TI CC2531 startup

nicolaiw, 2019-04-05:
Avoid while (true) { .. } during TI CC2531 startup

see:
https://github.com/zigbeenet/ZigbeeNet/blob/master/src/ZigBeeNet.Hardware.TI.CC2531/ZigBeeDongleTiCc2531.cs#L231

nicolaiw, 2019-04-05:
Because there is already an event the network state (see: https://github.com/zigbeenet/ZigbeeNet/blob/master/src/ZigBeeNet.Hardware.TI.CC2531/Network/NetworkManager.cs#L102) it should not be too difficult to refactor the Startup() method of the ZigBeeDongleTiCc2531 class returning a Task. For this I would recommend a TaskCompletionSource where the result will be set in the event handler.

Mr-Markus, 2019-06-11:
I began work on branch https://github.com/zigbeenet/ZigbeeNet/tree/feature/asyncStartup_issue46 and most code is implemented, but there is one thing which does not work.

When I start the sample application the hardware sends first requests sucessfully and gets responses, but while calling CreateEndpoint method and sending an AFRegister Request the response is null: https://github.com/zigbeenet/ZigbeeNet/blob/e005f2f4898742b486e577a7abc216ccd3403d15/libraries/ZigBeeNet.Hardware.TI.CC2531/ZigBeeDongleTiCc2531.cs#L321-L325

Does anyone have an idea?

I will fix this issue before creating the first release, because it is an breaking change for the caller of the library

Mr-Markus, 2019-11-29:
Closed due to inactivity. Does not break programm

nicolaiw, 2020-01-23:
Hello Markus,

would you mind to reopen this Issue ? Just as a reminder.

regards,
Nicolai