Mr-Markus/ZigbeeNet #47: Tested devices

Mr-Markus, 2019-04-13:
We need your help!

Do you already have tested some devices with ZigbeeNet? Please let us know which devices and which clusters or functionality you have tested so that we can create a "supported devices" section in our README

Thanks!

thombrink, 2019-04-13:
I got my [Ikea Tradfri lamp](https://www.ikea.com/nl/nl/catalog/products/60338452/) working in combination with my CC2531 usb stick.

Mr-Markus, 2019-04-13:
I have a **CC2531** usb stick and got Following devices working:

- [Ikea E14 Tradfri lamp](https://www.ikea.com/nl/nl/catalog/products/60365271/) (On/Off, Dimming)
- [Osram Smart+ Plug](https://www.amazon.de/dp/B074PZLX2P/ref=cm_sw_r_tw_dp_U_x_HHASCbT7NDK2C) (On/Off)
- [Osram Smart+ E27 Color LED](https://www.amazon.de/dp/B074PZSW1Y/ref=cm_sw_r_tw_dp_U_x_jJASCbQYCHXKM) (On/Off, Dimming, Change Color)

sixty-four-bit, 2019-05-01:
Hi @Mr-Markus   i have a CC2531 device . 
But it does not show as a com port.

I have installed the drivers. It looks ok on device manager and sniffer tool works with it .
But what ever i do it does not show as com port. 

Any suggestion.

Mr-Markus, 2019-05-01:
Hi @sixty-four-bit,
thanks for your interest in our library.

You have to flash the ZNP (ZigbeeNetworkProcessor) firmware from the Z-Stack directory here:
***\Z-Stack 3.*\Projects\zstack\ZNP\CC253x\dev\CC2531ZNP-with-SBL.hex**

Are you able to flash it? You will need CCDebugger hardware to do it.

Mr-Markus, 2019-05-05:
@andreasfedermann 
Which devices did you have used until now with your XBee hardware?

andreasfedermann, 2019-05-07:
**Coordinator**
Digi Xbee
Product Family: XB24C S2C
Function Set: ZIGBEE TH Reg
Firmware: 4060

**Smart Devices**
- [Philips hue light bulb white ambiance](https://www2.meethue.com/en-us/p/hue-white-ambiance-doublepack-br30/046677530167) (On/Off, Dimming, Change Color)
- [hue white and color ambiance go portable light](https://www2.meethue.com/en-us/p/hue-white-and-color-ambiance-go-portable-light/714606048) (in progress)
- [hue white ambiance fair ceiling light](https://www2.meethue.com/en-us/p/hue-white-ambiance-fair-ceiling-light/4100248U7) (in progress)

bmachuletz, 2019-07-10:
Hi,
I've watching this project since the beginning. I am coding a .NET Smarthome-Broker (it's currently not ready for the public). Nowaday I am using Zigbee2Mqtt to integrate Zigbee-Devices, but I am looking forward to integrate ZigbeeNet. I did some tests with my XIAOMI-(Temperature, Humidity, Pressure) Sensor and  I am able to get the values as expected.
Now my Question: What is the correct way to natively support new devices in ZigbeeNet <- I don't know where to start in the code (which classes have to be extended).
Maybe you can give me a fast intro on how to support new devices (I can test Tradfri, Heiman Smoke detector and Xiami Temperature Sensors at this time).

Mr-Markus, 2019-07-10:
Hi @bmachuletz ,

thanks for your interest in our library. That sounds great! I will be very interested in your project and looking forward to it.

I have wrote some articles in our wiki page. 

[Getting Started](https://github.com/zigbeenet/ZigbeeNet/wiki/Getting-started)
[Get node and endpoint address](https://github.com/zigbeenet/ZigbeeNet/wiki/Get-node-and-endpoint-address)
[Read attributes](https://github.com/zigbeenet/ZigbeeNet/wiki/Read-attributes)
[How to control devices](https://github.com/zigbeenet/ZigbeeNet/wiki/How-to-control-devices)

You can also take a look at the [Playground Demo project](https://github.com/zigbeenet/ZigbeeNet/blob/develop/samples/ZigBeeNet.PlayGround/Program.cs)

If you need further information I will help you whenever you need it. Just open an new issue for it.

Mr-Markus, 2019-08-20:
If there are any new devices you have tested just open a new issue

closed 2019-08-20 by Mr-Markus

victornor, 2022-11-13:
I can confirm that the <a href="https://www.aqara.com/us/motion_sensor.html">Aqara Occupancy and Illuminance sensor (RTCGQ11LM)</a> is working using the IlluminanceLevelSensingCluster and the OccupancySensingCluster.