Mr-Markus/ZigbeeNet #49: Avoid while (!_closeHandler) { ... } in XBeeFrameHandler.Start()

nicolaiw, 2019-05-06:
This might be resolved with a TaskCompletionSource.

Here is an example of a refactored while loop with a TaskCompletionSource:
   https://github.com/zigbeenet/ZigbeeNet/blob/master/src/ZigBeeNet/Transaction/ZigBeeTransaction.cs#L82

andreasfedermann, 2019-05-07:
Hello @nicolaiw,

I will refactor this as soon as possible. I think there are some more refactoring actions necessary. Some code is not very c# style I think.
First of all I will check all existing while loops and refactor this if appropriate.

nicolaiw, 2019-05-08:
This is not jus an Issue for Xbee 

[image]

This is the CPU usage of the PlayGround app with a TIC2531 while waiting for input. So you are right .. removing or avoid while loops where ever we can should be out next Task :D

nicolaiw, 2019-05-20:
[image]

I could reduce the CPU usage (same machine) by refactor the while loop which reads from the serial port. The previous versions always checked for BytesToRead in a while loop which is not necessary and very inefficient. The refactored version waits until data available through ReadByte(). After that it reads the rest of the data. I think I could be done even more elegant. Instead of using ReadByte() to wait for data we might use the SerialPort.Read(byte[] buffer, int offset, int count) methode to read the header, parse it for the payload length and read the payload. I'm not sure but this might be Hardware/Dongle dependent (size of the header). I guess this won't be decrease the cpu usage but might be more elegant.

About 34% CPU usage still seems too high. I think there is another loop or something like that which utilizes the CPU.

@Mr-Markus @andreasfedermann 
Maybe you could help me to find those places in the code.

@andreasfedermann 
Could you please pull the latest version and try out if this version still works with your Xbee hardware ?

Best regards

nicolaiw, 2019-05-20:
Found one:

https://github.com/zigbeenet/ZigbeeNet/blob/master/src/ZigBeeNet.Transport.SerialPort/ZigBeeSerialPort.cs#L157

nicolaiw, 2019-05-20:
[image]

Looks much better to me and thats what I would expect when there is no data to read :)

best regards

andreasfedermann, 2019-06-02:
Hi Nicolai,

sorry for being offline that long but I’ve moved. So I had less time to write some code. My aim for this week is to test your code changes. 

Best regards!

Mr-Markus, 2019-06-02:
Hi @andreasfedermann,
good luck with your new home. I hope you will use this library for your smart home solution there :-)

Best,
Markus