Mr-Markus/ZigbeeNet #52: Refactor ZCL autocoder to use xml instead of md

nicolaiw, 2019-05-15:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/commit/5667ab1d7e2de610043bb5158e5a576f219039ec

spudwebb, 2019-11-15:
any progress with this?
It looks like the zcl_defintion.md is missing a lot of attributes, for example the door Lock, and Fan Control don't have any. The xml files look much more complete.

nicolaiw, 2019-11-16:
Hi spudwebb,

first at all .. thank you for your PR's to this repository!

There is already a branch for #50 and #52 (branch Issue_50_and_52). Currently I do not have the time to finish this work.

regards

spudwebb, 2019-11-18:
I will give it a try, but could you give me a quick state of play of this branch? What is done and what remains to be done?
thanks

TheBlubb14, 2019-11-18:
Hello, everybody. In my spare time I also started to generate code from the .xml files. I wanted to get away from the StringBuilder and started to work into the C# Roslyn code generation. I have also come relatively far. If I'm not mistaken, all I need is the serialize and deserialize method. Unfortunately I haven't made any further progress in the last few weeks. Currently I have my test project under https://github.com/TheBlubb14/RoslynCodeGeneratorTest

nicolaiw, 2019-11-18:
Hello guys,
in the Issue_50_and_52 branch files are getting generated but it's java as far as I remember.

spudwebb, 2019-11-22:
As I'm not familiar with Roslyn, and as I think it's a good idea to keep the code as close as possible as the java code, I took the Issue_50_and_52 branch as my starting point rather than the code from @TheBlubb14  (Thanks anyways!)
It is a lot more work than I thought, but I'm almost done. I will issue a PR when it's ready.

Mr-Markus, 2019-11-23:
Thx @spudwebb

closed 2019-11-23 by Mr-Markus