Mr-Markus/ZigbeeNet #56: Potential license problem

Gallimathias, 2019-05-28:
Since the project is a port of zsmartsystems, which is equivalent to a copy in the broadest sense, the copyright and also the license must be respected.

zsmartsystems/com.zsmartsystems.zigbee uses the EPL-1.0.

This is a CopyLeft license so the notes as well as the license of the original project have to be taken over. MIT in this form is not allowed.

I recommend a quick customization so you don't get into legal trouble as the owner ;)

nicolaiw, 2019-05-28:
Hey @Gallimathias,

any suggestions ?

Thank you for pointing this out.

Best regards

Gallimathias, 2019-05-28:
The license says it all

MIT must be changed to the same [license as the original project](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/LICENSE)

And we have to respect this comment from the project:

> Some parts of this code are from zigbee4java which in turn is derived from ZB4O projects which are licensed under the Apache-2 license. These are being refactored out to ensure the contributions to this reportisory are understood.

And potentially the copyright notices of the original project to the Zigbee documentation must be taken into account if we use the same documents here.

> Some documentation used to create parts of this framework is copyright © ZigBee Alliance, Inc. All rights Reserved. The following copyright notice is copied from the ZigBee documentation.
> 
> Elements of ZigBee Alliance specifications may be subject to third party intellectual property rights, including without limitation, patent, copyright or trademark rights (such a third party may or may not be a member of ZigBee). ZigBee is not responsible and shall not be held responsible in any manner for identifying or failing to identify any or all such third party intellectual property rights.
> 
> No right to use any ZigBee name, logo or trademark is conferred herein. Use of any ZigBee name, logo or trademark requires membership in the ZigBee Alliance and compliance with the ZigBee Logo and Trademark Policy and related ZigBee policies.
> 
> This document and the information contained herein are provided on an “AS IS” basis and ZigBee DISCLAIMS ALL WARRANTIES EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO (A) ANY WARRANTY THAT THE USE OF THE INFORMATION HEREIN WILL NOT INFRINGE ANY RIGHTS OF THIRD PARTIES (INCLUDING WITHOUT LIMITATION ANY INTELLECTUAL PROPERTY RIGHTS INCLUDING PATENT, COPYRIGHT OR TRADEMARK RIGHTS) OR (B) ANY IMPLIED WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE, TITLE OR NONINFRINGEMENT. IN NO EVENT WILL ZIGBEE BE LIABLE FOR ANY LOSS OF PROFITS, LOSS OF BUSINESS, LOSS OF USE OF DATA, INTERRUPTION OF BUSINESS, OR FOR ANY OTHER DIRECT, INDIRECT, SPECIAL OR EXEMPLARY, INCIDENTIAL, PUNITIVE OR CONSEQUENTIAL DAMAGES OF ANY KIND, IN CONTRACT OR IN TORT, IN CONNECTION WITH THIS DOCUMENT OR THE INFORMATION CONTAINED HEREIN, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH LOSS OR DAMAGE.
> 
> All Company, brand and product names may be trademarks that are the sole property of their respective owners.

And

> Dongle Documentation
> Some documentation used to implement dongle drivers is copywrite to the respective holders, including Silicon Labs, Texas Instruments, Dresden Electronics, Digi International.

Mr-Markus, 2019-05-28:
It would be very helpful for us if you can create a PR with all necessary text and stuff.

Docs from Texas are not included in this repository, but only from zigbee alliance which are availible on their website. A copyright text as you said seems to be important, too.

closed 2019-05-29 by Mr-Markus