Mr-Markus/ZigbeeNet #61: Change project workflow

Gallimathias, 2019-05-30:
To make the work with each other easier I suggest to get closer to GitFlow for this project.

At least a separation of master (deployed) and develop (current alpha) is useful and advisable. Especially with regard to issue #33 and possible auto builds of the nuget packages.

Mr-Markus, 2019-06-03:
I like the idea and would appreciate to work this way in the future.

In the past there were many changes and there was no version to release, but now we should be able to create a first alpha version

@nicolaiw 
What do you think?

andreasfedermann, 2019-06-03:
We should also link the test label from appveyor in the readme. So everybody could see our test status.

Mr-Markus, 2019-06-03:
@andreasfedermann 
Do you have a screenshot? I can't find any other badges in appvayor settings. is it an external badge plugin?

Mr-Markus, 2019-06-03:
I created a develop branch with git flow init -d

@Gallimathias Did I anything forgot?

From now we can should work with git-flow

Thanks for this issue :-)

Gallimathias, 2019-06-03:
The git flow only affects each user locally.

Actually, this is simply a way of working that avoids conflicts and makes working in a team easier.

Therefore the structure should be as follows 

master -> published version
develop -> current development
feature -> feature to be developed

In my opinion the shortest and most effective cheatscheet for git flow by Daniel Kummer:

[English](https://danielkummer.github.io/git-flow-cheatsheet/)
[Deutsch](https://danielkummer.github.io/git-flow-cheatsheet/index.de_DE.html)

Mr-Markus, 2019-06-03:
Yeah, it is really simple and working in teams is much easier.
I have used TFVC in the past with TFS Server @work, before we switched to git in our company. This was a huge step and change for me and our team. Then we did branching manually, but from now I will use git-flow because it is much easier :-)

Sometimes you use something without knowing that it has a name or existing tools 😃

@Gallimathias 
Do you have a suggestion about appvayor configuration? I already thought about creating nuget packages from master branch automatically

Gallimathias, 2019-06-03:
Mhm actually tends to fit the big picture first. Of course it would be practical to create the artifacts. Automated Nuget packs can be a good thing. We can limit it in the script to the master branch so every release would be deployed automatically. You might want to think about whether you want a big Nuget Package or separate Nuget Packages e.g. to separate the hardware implementations and the core.

I'm going to open a new issue

Translated with www.DeepL.com/Translator

closed 2019-06-04 by Mr-Markus