Mr-Markus/ZigbeeNet #63: ConBee II support?

jgmdavies, 2019-06-13:
Hi All,

Does the current XBee support work for a ConBee II dongle please?

Thanks,
Jim

Mr-Markus, 2019-06-13:
Hi @jgmdavies,
we have no Conbee support at the moment. But our library is open for it if somebody will implement it.

As far as I know XBee is from Digi and Conbee from Dresden Elektronik, so there are two implementations which are different. Do you have heard about a relation between these two manufacturers?

jgmdavies, 2019-06-13:
Thanks Markus.

You're right of course, and no, I don't know of a relationship - I'm afraid I was being over-optimistic!

DavidKarlas, 2019-12-31:
@jgmdavies FYI, there is PR now open that adds support for ConBee https://github.com/Mr-Markus/ZigbeeNet/pull/94

Mr-Markus, 2019-12-31:
Great!

jacobussystems, 2019-12-31:
Fantastic - many thanks for alerting me to this David!
Jim

Mr-Markus, 2020-01-19:
@DavidKarlas 
Are you still working on this?

DavidKarlas, 2020-01-19:
No, but its working as is. I'm not aware of any bug, so I guess this can be closed.

Mr-Markus, 2020-01-19:
Great. I didn't dive deep in this issue yet, but do you think it should be possible to control other devices, too?
In my opinion it should not matter if you control light or other devices if the ZigBee ZCL packet is correctly sending and receiving over serial port.

DavidKarlas, 2020-01-19:
I don't see why, other devices wouldn't work. As you said, since ZigBee ZCL packet are flowing correctly... Everything should just work...

Mr-Markus, 2020-01-19:
Well done. This was a great improvment and PR from you. Many thanks. I am very happy about this and our third hardware implementation 😊

closed 2020-01-19 by Mr-Markus