Mr-Markus/ZigbeeNet #65: get device information from IKEA LED1732G11

bmachuletz, 2019-07-15:
Hi,
is anyone able to help me to get device Information (model) from that type of bulb?
The "manufacturer id" is already discovered but i cant get more Information.

thx

bmachuletz, 2019-07-16:
ok got it:)

First
-----
`networkManager.Send(test, new ReadAttributesCommand { Identifiers = new List<ushort> { ZclBasicCluster.ATTR_MODELIDENTIFIER, ZclBasicCluster.ATTR_MANUFACTURERNAME } });`

Second
---------
In an IZigbeeCommandListener set a filter like this:
```
if (command is ZigBeeNet.ZCL.Clusters.General.ReadAttributesResponse)
{
              ((ZigBeeNet.ZCL.Clusters.General.ReadAttributesResponse)command).Records.ForEach(record =>
                {
                    
                    Console.WriteLine(record.AttributeValue);
                });  
}
```

closed 2019-07-16 by bmachuletz

Mr-Markus, 2019-07-25:
Hi @bmachuletz ,

you also able to retrieve the model name like following:

```cs
ZigBeeNode node = _networkManager.GetNode(4711);
string model = node.Endpoints.FirstOrDefault().Value.GetInputCluster(ZclBasicCluster.CLUSTER_ID)).GetModelIdentifier(60); // caches value for 60 secods
```

You can do this with any cluster and attribute 😉

Mr-Markus, 2019-07-25:
I will retrieve something like this here:

[image]

bmachuletz, 2019-07-27:
Hello Markus,sorry for the late response. I am in holiday mode.In this case you will get the string „Tradfri bulb...“.How can I do this with a different attribute, in order to get the string „LED ...“?