Mr-Markus/ZigbeeNet #66: manufacturer / model specific extensions of autogenerated code

bmachuletz, 2019-07-20:
Does it make sense, for example, to expand the autogenerated ZclCluster classes in derived classes?
Currently I am testing with an Ikea-Tradfi bulb. Here the imprinted Model Id is stored in the ZclBasic cluster under the vendor-specific attribute 0x000a. Currently I am expanding the cluster in my own code.

Mr-Markus, 2019-07-20:
How should the structure be? I Don't know what you have implemented

The base class is ZclCluster and each cluster has its own class which inherits from that class.

The manufacturer model ID should be in the same cluster for all manufacturers

bmachuletz, 2019-07-20:
As far is I‘ve read in the Zigbee documentation the 0x000a Attribute is not mentioned.
How do you think it‘s best to Integrator into ZigbeeNet. Can it be a standard attribute  in the ZclBasicCluster?

Mr-Markus, 2019-07-21:
Why do you need the attribute 0x000a?

In your question in #65 you have used 0x0005 for model identifier.

Also see here https://github.com/zigbeenet/ZigbeeNet/blob/master/autocode/ZigBeeNet.CodeGenerator/Resources/zcl_definition.md#attributes

If there are manufacturer specific extensions it will be an inherited cluster. See 2.2.1.2 and 2.3.3 in [ZCL docs](https://github.com/zigbeenet/ZigbeeNet/blob/master/docs/zigbee-protocol/Zigbee-cluster-library-specification.pdf)

Currently we do not have any of them but I think it should be possible by adding them to the Autocode Ressource file mentioned above 

Do you have something we should add?

bmachuletz, 2019-07-21:
Hi,
Ikea saves the model id in 0x000a and it is reported („LED1732G11“) as a character  string.
Yesterday I took a look at the autocoder I can edit the resource file in my own copy and check whether it works as expected or not. I think it should:) Also I can make a pull request after all the tests.
Greetz Benjamin.

Mr-Markus, 2019-07-21:
Where did you got the information? It will be interesting for me, too.
I am not sure if it is correct to extend the ZCL basic cluster if it is manufacturer specific. It would be more meaningful to create one derived class or cluster for each manufacturer and cluster.

Because ZCL clusters are defined by the zigbee alliance I would prefer to keep it's structure in our library

bmachuletz, 2019-07-21:
I read about it in this thread: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1032
And sorry, it‘s not a character string. It is an octet string.

nicolaiw, 2019-07-22:
Hello @bmachuletz,

there is still work in progress in branche Issue_50_and_52. This branche refactores the autocoder to support manufacturer-specific clusters, commands, and attributes https://github.com/zigbeenet/ZigbeeNet/issues/50. There is a Project called ZigBeeNet.CodeGenerator which already generates files but java. I think we just have to make it create c#. Feel free to check out this branche and make it create c# code :D

Best regards
Nicolai

nicolaiw, 2019-07-31:
Hello guys,

if someone wants to help: I continued working on c# code generation (ZclCluster- and ZclCommand- classes). When running the ZigBeeNet.CodeGenerator project the files will currently stored at C:\temp\csharp.

Best regards
Nicolai

bmachuletz, 2019-08-01:
Hello Nicolai,I had a short look on your code. When I am back from my holiday I would take a closer look on it. Thank you very much!

nicolaiw, 2019-08-01:
Hello @bmachuletz,

have a nice vacation :)

Best regards
Nicolai

Mr-Markus, 2019-11-25:
Hi @spudwebb 
is this also fixed with your PR #82 ?

spudwebb, 2019-11-25:
I haven't tested it but yes you can now create a manufacturer specific cluster, or derive an existing cluster to add manufacturer specific attributes. The new cluster class just need to overrides:

`public virtual int? GetManufacturerCode()`

or the derived cluster class needs to create the new attributes with the manufacturerCode set:

```
public ZclAttribute(ZclCluster cluster, ushort id, string name, ZclDataType dataType,
                bool mandatory, bool readable, bool writable, bool reportable, int manufacturerCode)
```

nicolaiw, 2019-11-25:
@spudwebb Thank you for your contribution!

closed 2019-11-26 by Mr-Markus