Mr-Markus/ZigbeeNet #67: Support for Ember stack

spudwebb, 2019-09-20:
Any plan or anyone currently working on supporting Ember stack?

I may need this in a near future, so I wanted to ask before I start trying to add it myself.

Mr-Markus, 2019-09-21:
Hi,
Currently nobody is working on an ember Support.
We will be happy to get a PR from you with it.

If you do have any questions do not hesitate in contacting me

Markus

spudwebb, 2019-10-04:
I changed my plans, I bought a Digi XStick (XU-Z11) and I'm testing the library with it.
After a fix for baud rate (looks like the XStick needs 9800) I have it communicating with the Playground console app.

Now, I want to add/control a Philips Hue bulb. Can you tell me how to do that?

Sorry for the noob question, I'm new to Zigbee!

Mr-Markus, 2019-10-05:
Hi there,
No problem, you can ask everything you need. I am happy that you are interested in this library and hope I can help you. 

If you have started the Playground app you can enter Port and baudrate with selected dongle. Than the dongle should be startup. After that you should be able to enter "join" to allow zigbee devices like your bulb to join the network. After entering "join" you need to get the bulb in the join mode. For that just Google how to do this. I know that Osram bulbs needs to switch on and off 5 times for example. With Philips hue I don't know it exactly how to do this, but @andreasfedermann have done this before. Maybe you know it?

I use the Texas Instrument CC2531 dongle, so that I can not test it, but @andreasfedermann have implemented the Xbee dongle and should know further information.

I hope that helps you. Otherwise don't hesitate in contact me 

I am also working on a Rest Api which uses ZigbeeNet. Maybe it will be help you too. See [here](https://github.com/Mr-Markus/ZigbeeNet-webapi)

spudwebb, 2019-10-08:
Hi,
Thank you for your answers.
I have added my Hue bulb, and I can now control it using the Playground program.  Yihaa!
Here is more questions:

(1) If I understand this correctly the nodes information are not stored on the dongle, so we have to store them somewhere using JsonNetworkSerializer, correct?
If so why the following line was commented out in the Playground project?

 `networkManager.NetworkStateSerializer = deviceSerializer;`

this seems to work if I uncomment it, but I want to make sure there isn't any known problem with it?

(2) how do I now what commands are supported by a node?

(3) how do I get the current status (on/off, level, color, etc...) of a node?

(4) have you ever tested ZigBeeNet with Mono?

To give you a little more context, my goal is to build a Zigbee driver for Homeseer

thanks!

Mr-Markus, 2019-10-08:
Hi,

great news! I appriciate your project and will look at it with much interest.

Here are answers to your questions

> (1) If I understand this correctly the nodes information are not stored on the dongle, so we have to store them somewhere using JsonNetworkSerializer, correct?
If so why the following line was commented out in the Playground project?

That's correct that paired nodes are not stored on the dongle. Maybe it can with neigbour tables, but I don't know that exactly. Maybe it is internally only. But for a broker software or home server you need to store much more information about nodes, so you need to store it to a own database like JSON file or database like mongodb, ms sql, or whatever.
Just for an example there is a Implementation of the `IZigBeeNetworkStateSerializer` interface that handles it for saving to a file. If you set a NetworkStateSerializer object that implements this class it will automatically serialize a new node on adding it to the network and also deserialize all nodes on dongle startup and adds them to the network.
Just implement your own data store with that interface

> (2) how do I now what commands are supported by a node?

You need to store some information in your database after adding the node. For that you have to implement own logic that will read values from ZigBeeNode object instance which was added. The best way is to implent a own class which implements the IZigBeeNetworkNodeListener interface, add it to the network manager and do this here. It will be executed when a node was added to the network.
Most useful information is which clusters the node owns. E.g. if node have OnOff Cluster you know that you can use the ZclOnOffCluster and it's methods.

> (3) how do I get the current status (on/off, level, color, etc...) of a node?

Just take a look at our wiki [here](https://github.com/Mr-Markus/ZigbeeNet/wiki/Read-attributes) which should describe the concept how to read attributes. You can replace BasicCluster with ZclOnOffCluster or ZclColorControlCluster etc. and will see the Methods for reading it's availible attributes

> (4) have you ever tested ZigBeeNet with Mono?

Yes and it works very well. But there is a better solution which I have implemented today. Since a few days there is netcore 3.0 released which now allows netcore to run System.IO.SerialPort directly on linux. See [here](https://docs.microsoft.com/de-de/dotnet/core/whats-new/dotnet-core-3-0#serialport-for-linux)

So I updated the necessary projects to net core 3.0 in this 4c4e1f0c522282490e42fe19309664ccc0b340f2 commit and tested it sucessfully on ubuntu
Maybe that will be a solution for you, too

spudwebb, 2019-10-09:
thank you for your answers

That's good news about Mono, I won't be able to use net core 3.0 because Homeseer runs using Mono on Linux

Mr-Markus, 2019-10-09:
You're welcome. Feel free to open another issue if you have any questions

closed 2019-10-09 by Mr-Markus