Mr-Markus/ZigbeeNet #70: NodeUpdated never called

spudwebb, 2019-10-16:
When a node is first added  NodeAdded is called, but the node passed as parameter has an empty Endpoints dictionary, which I guess is expected?

Then  ZigBeeDiscoveryExtension does its job and queries the node for its capabilities. The Endpoints  dictionary is populated, but NodeUpdated is never called.

I think ZigBeeNetworkManager.RefreshNode() is supposed to trigger the NodeUpdated callback for all the listeners but it never does because it always returns here:

```
               // Return if there were no updates
                if (!currentNode.UpdateNode(node))
                {
                    Log.Debug("{IeeeAddress}: Node {NwkAddress} is not updated", node.IeeeAddress, node.NetworkAddress);
                    return;
                }
```

I don't understand how it is supposed to work: currentNode.UpdateNode(node) always returns false because here currentNode and node are the same reference so UpdateNode correctly says they are equal.

I'm testing with the PlayGround sample

spudwebb, 2019-10-16:
I think we need to port:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/671
and probably also:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/685

Mr-Markus, 2019-10-19:
I understand what you mean. Maybe that is the reason why sometimes new devices are not added correctly to the network after restarting the dongle.
But I am not sure at the moment and will take a loot at this point of code.

Also the two linked issues will make sense to implement them. I will also read about them and will open new issues if they need to be implemented

spudwebb, 2019-10-22:
Do you want me to port this 2 PR to ZigbeeNet?

Mr-Markus, 2019-10-22:
This would be very helpful for everyone. 

I would be very happy about your PR :-)

spudwebb, 2019-10-23:
done in PR #71

closed 2019-10-23 by Mr-Markus