Mr-Markus/ZigbeeNet #74: Attribute reporting

spudwebb, 2019-10-26:
I'm trying to configure attribute reporting on a light bulb, so that the value is automatically reported to my program when on/off status change or every 10 seconds:

```
var cluster = endpoint.GetInputCluster(ZclOnOffCluster.CLUSTER_ID);
if (cluster != null)
{
	var result = await cluster.Bind(networkManager.LocalIeeeAddress, 1);

	if (result.IsSuccess())
	{
		result = await ((ZclOnOffCluster)cluster).SetOnOffReporting(1, 10);

		if (result.IsSuccess())
		{
		}
	}
}
```

but I always get the following response:

[11:21:46 DBG] RX APS: ZigBeeApsFrame [sourceAddress=20978/11, destinationAddress=0/1, profile=260, cluster=6, addressMode=0, radius=0, apsCounter=0, payload=24 14 7 140 0 0 0]
[11:21:46 DBG] RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=False, direction=SERVER_TO_CLIENT, disableDefaultResponse=True, manufacturerCode=0, sequenceNumber=14, commandId=7]
[11:21:46 DBG] RX CMD: ConfigureReportingResponse [On/Off: 20978/11 -> 0/1, cluster=6, transId=14, Status=SUCCESS, Records=[Attribute Status Record: **status=UNREPORTABLE_ATTRIBUTE**, direction=False, attributeIdentifier=0,]]

first I don't know if I need to bind? I'm not clear what a bind request does and when I should send it?
I tried with or without bind, I tried with different attributes but I always get status=UNREPORTABLE_ATTRIBUTE

any idea?  have you ever  tested reporting?

Note: see my PR #77 for some fixes needed for bind and config reporting command

Mr-Markus, 2019-10-26:
I have no idea about binding, sorry. When I began this project I thought that I need to bind a device to send and receive data, but then I realised that it is no necessary.

I read data from a device by using an endpoints cluster methods. I described it here:
https://github.com/Mr-Markus/ZigbeeNet/wiki/Read-attributes

But I think it is not a good idea to read data every few seconds (polling), but believe that there is something like binding, that notifies the coordinator or our programm about value changing.

Maybe you can found more information in the official Zigbee docs here:
https://github.com/Mr-Markus/ZigbeeNet/tree/develop/docs/zigbee-protocol

spudwebb, 2019-10-28:
> But I think it is not a good idea to read data every few seconds (polling), but believe that there is something like binding, that notifies the coordinator or our programm about value changing.

yes this done by using the ZclCluster.SetReporting method to configure what attribute should be reported and when.

after some googling it looks like the Hue bulb that I'm testing with do not handle this feature, so this is probably why I get the UNREPORTABLE_ATTRIBUTE status. So I guess I need to get some other zigbee devices to test with.

if you can test this feature with the devices you have, that would be great.

Mr-Markus, 2019-10-28:
I own Hue, Ikea and Osram devices and will test it with them. Is it enough to use your code below to do this? And should the reported attributes be displayed in the console logging when it toggles on/off for example? Just that I know what I should test.

`after some googling it looks like the Hue bulb that I'm testing with do not handle this feature, so this is probably why I get the UNREPORTABLE_ATTRIBUTE status. So I guess I need to get some other zigbee devices to test with.`
Yes, I learned that some manufacturers do not support some special features. But it is documented in official Zigbee docs: https://zigbee.org/zigbee-products-2/

Then take a look on the documents, where the manufacturers answered some common questions (e.g. if they implement an attribute)
[image]

Mr-Markus, 2019-11-19:
Do you have further questions or can this issue be closed?

spudwebb, 2019-11-19:
Did you had a chance to test reporting on your devices? If so does it work?

Mr-Markus, 2019-11-20:
At the moment I have only Osram bulbs for testing. Also I am moving within next weeks, so that my equipment and time is limited. But if I have the chance to test it I will comment it here

spudwebb, 2020-10-13:
Attribute reporting works fine. Closing this issue

closed 2020-10-13 by spudwebb