Mr-Markus/ZigbeeNet #75: Make ToString method of Request and Response handle List

spudwebb, 2019-10-28:
currently most ZDO and ZCL request/response do not handle well properties that are list. They rely on the  Object.ToString() method that just print the type of the list:

[14:21:10 DBG] RX CMD: NetworkAddressResponse [20978/0 -> 0/0, cluster=32768, transId=, status=SUCCESS, ieeeAddrRemoteDev=17880104E15812, nwkAddrRemoteDev=20978, startIndex=0, nwkAddrAssocDevList=**System.Collections.Generic.List`1[System.Int32]]**

we need to see the content of the list

For ZCL that means the code generator need to be modified to handle lists

Mr-Markus, 2019-10-28:
Yes, that will be a helpful enhancment. I would tag this for implementing

Mr-Markus, 2019-10-28:
@nicolaiw 
maybe something for you? 😁

spudwebb, 2019-11-26:
The change above fix this for ZCL, and  ZDO will be automatically fixed too once #84 is done

spudwebb, 2020-10-13:
I believe this is now ok for both ZDO and ZCL

closed 2020-10-13 by spudwebb