Mr-Markus/ZigbeeNet #81: avoid return await

nicolaiw, 2019-11-18:
Like here https://github.com/Mr-Markus/ZigbeeNet/blob/develop/libraries/ZigBeeNet/ZigBeeNetworkManager.cs#L992 we should avoid to call "return await" if possible as the compile will generate a state machine. Just return the Task direcly. In most cases it is not necessary to call await when there is nothing to do after awaiting the task. (It might by when combined with using)

nicolaiw, 2019-11-18:
I made a quick example. Have a look at the difference IL code.

https://sharplab.io/#v2:CYLg1APgAgTAjAWAFBQAwAIpwKwG5nJQDMmM6AwugN7Lp3q30AOATgJYBuAhgC4CmmAGzoASgFcAdgEEAzgE8JAYwD6cABQBKRnRpJ6+zAHYhAOnES1m9AF4AfNQC+G/HvoPtDV3VadeAqAAcQqKSsgoqMJoeugb0UACcpuaWGjb2VE4u+u5IDkA

spudwebb, 2019-12-13:
fixed in PR #90

Mr-Markus, 2019-12-13:
Thx for quick implementation

closed 2019-12-13 by Mr-Markus