Mr-Markus/ZigbeeNet #89: Use Task.Run() instead of Start()

nicolaiw, 2019-12-13:
https://github.com/Mr-Markus/ZigbeeNet/blob/065fc7f402efd82a3333155edc756214caade1a3/libraries/ZigBeeNet/Database/ZigBeeNetworkDatabaseManager.cs#L164

see: https://stackoverflow.com/questions/29693362/regarding-usage-of-task-start-task-run-and-task-factory-startnew

spudwebb, 2019-12-13:
Task.Run does not allow to pass TaskCreationOptions.LongRunning, so shouldn't we use Task.Factory.StartNew instead?

also there is the same problem here:
https://github.com/Mr-Markus/ZigbeeNet/blob/065fc7f402efd82a3333155edc756214caade1a3/libraries/ZigBeeNet.Transport.SerialPort/ZigBeeSerialPort.cs#L99-L102

nicolaiw, 2019-12-13:
ur right .. this should be changed as well .. I would agree .. StartNew() would be an option .. another topic is the call to Task.Result in this project .. this should be avoided ..

spudwebb, 2019-12-13:
fixed in PR #90

Mr-Markus, 2019-12-13:
Good improvement

closed 2019-12-13 by Mr-Markus