Mr-Markus/ZigbeeNet #91: Dongle Support for Nortek Security Control

BenLiyanage, 2019-12-29:
I have a Nortek Security Control (https://www.amazon.com/GoControl-CECOMINOD016164-HUSBZB-1-USB-Hub/dp/B01GJ826F8/ref=asc_df_B01GJ826F8/?tag=hyprod-20&linkCode=df0&hvadid=193989831776&hvpos=1o1&hvnetw=g&hvrand=11729360482103299148&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9031961&hvtargid=pla-312424172362&psc=1).  What do I need to get this running?  Can I just treat this as a serial port? I have some programming experience if we need to build some sort of adapter for this, though I'd probably need a little help to get started as I don't do much hardware programming.

BenLiyanage, 2019-12-29:
Worth mentioning is it appears to show up as a com port:

[image]

Mr-Markus, 2019-12-29:
Hi there,
Thanks for your interest in ZigBeeNet. 
I heard of this Hardware the first time. I found out that it has an Ember Chip inside. This is currently not supported, but technically possible. But you need deeper C# skills for implementing.
I would recommend you to buy a Texas Instrument CC2531 USB stick, which is fully implemented. It costs about $8 on ebay or amazon

BenLiyanage, 2019-12-30:
I'm not shy of doing some C#.  This isn't my first rodeo.

BenLiyanage, 2019-12-30:
Probably a better question is with this library and a working dongle will I be able to join my computer to a Philips hue hub and receive light commands?  I have a bunch of RGB lights that I would like to drive via the Hue software.

Mr-Markus, 2019-12-31:
This library does not need a Philips Hue hub, but just a ZigBee Dongle. Currently there are Texas Instrument CC2531 and XBee supported.
Feel free to implement necessary Interfaces to support more hardware like your Ember Chip.
Just tanke a look at the existing projects to see what is needed. At first you need a NetworkManager class

BenLiyanage, 2019-12-31:
I think you missed my question. Will I be able to join my computer with the dongle to an existing zigbee network as a device rather then as a controller with this library?

Mr-Markus, 2019-12-31:
And what should your computer do if it joined to a network? Or which device type should it be?

At first you need a correct Firmware for your dongle. And if this firmware is able to be like a device it should no Problem to join the network

BenLiyanage, 2020-01-03:
I have RGB lights that I can control through my motherboard api. I’m looking to have my computer behave like an rgb light (or several) to the zigbee cluster.

BenLiyanage, 2020-01-03:
I was looking to take the api code from this project (https://github.com/tylerszabo/RGB-Fusion-Tool) and combine it with this zigbee library to create a service that would let it join my hue network. Then I was thinking about using the hue sync to have it drive color shifting based on what is on screen. 

If I can get my computer to show up/register as a light device I can wrangle all the code for the rgb settings. Does this seem viable with the xbee or ti chip?

Mr-Markus, 2020-01-07:
Interesting idea. I don't know it at the moment but will test it with two TI CC2531 dongles if they can communicate with the other.
If it works it should not matter which manufacturer dongle you will use.

Just give me a few days. I will reply here with my result

BenLiyanage, 2020-01-08:
Cool, I'm interested in hearing about your results!

Mr-Markus, 2020-01-29:
Ember Support with a Nortek hardware done in PR #95

Mr-Markus, 2020-04-08:
ZigBee specification describes that each network can only contain one coordinator. So it is Not possible that they talk to each other

closed 2020-04-08 by Mr-Markus