Mr-Markus/ZigbeeNet #99: Implement JSON and MongoDB DataStores

Mr-Markus, 2020-02-23:
I thought about implementing a MongoDB DataStore for another project that is using ZigBeeNet.

I implemented it in commit 6aaf764b4ec3864a59d3e710bc973068beaff900 and found out that it can be used out of the box with configuration parameters. So I created the library ZigBeeNet.DataStore.MongoDb that implements the  `IZigBeeNetworkDataStore` interface.

I referenced it in the PlayGround app in commit 7a591c93ed9a8cad72382c8c7f2b1efb6981b32d and made it configureable via command line args. I also extracted the JSON DataStore code to own library ZigBeeNet.DataStore.JSON so that it can be accessed to out of the box.

For some JSON serialization reasons I changed the `ZigBeeClusterDao` class in commit 679c5df53db17824d9783ac78954ed1b287bd586, so that it does not longer use Dictionary for attributes, but store them in a List. Because we can use Linq for accessing attributes we do not need a dictionary for finding a special attribute.

I think it is a little help for some developers that will start their own project and will access the network nodes in their application with these two adapters, because they can not only access them by ZigBeeNet NetworkManager class, but also are able to mix it with their code by connecting to that data source, too.

spudwebb, 2020-02-23:
> For some JSON serialization reasons I changed the ZigBeeClusterDao class in commit 679c5df, so that it does not longer use Dictionary for attributes, but store them in a List. Because we can use Linq for accessing attributes we do not need a dictionary for finding a special attribute.

What are the "JSON serialization reasons"?

Mr-Markus, 2020-02-23:
Attributes were stored with type `Dictionary<ushort, ZclAttributeDao>` and because that the key is not a string you have to configure how to serialize the dictionary, so that it can be correctly displayed in Json (maybe only in mongoDb, I don't know yet)

See here: https://mongodb.github.io/mongo-csharp-driver/2.10/reference/bson/mapping/
Section **Dictionary Serialization Options** near bottom

And I think that nobody is happy about complex queries when you use `ArrayOfArrays` as option, which creates a two dimensional array as key. I tested it and it is terrible. 
And because we can query the attributes by Linq I decided to change it the way I did.

And if you will use the JSON in e.g. JavaScript it will also be possible to query it by using .find() or .filter() methods

Any concerns?

spudwebb, 2020-02-24:
My only concern is that it is a breaking change. For the software I'm building, I'm going to need this kind of changes to be avoided as much as possible.

Mr-Markus, 2020-02-24:
Yes, breaking changes are always unpleasant. But sometimes they are necessary.

Do you have any suggestions how to solve this? maybe a inherited class?

Mr-Markus, 2020-03-11:
@spudwebb 
Did you found a solution for your software?

I thought about it again and  because it is just the Dao Object it should keep simple. So an List<> would be the best. Other Properties like Endpoints etc. were already a List<>
So it is now same implementation for everything.

If there are no further steps requeired i would close this issue. Next time I will create an issue before merging a feature into developer branch.

spudwebb, 2020-03-11:
My software is not released yet to the public so for now breaking changes can be overcome. You can close this issue.

closed 2020-03-11 by Mr-Markus