Mr-Markus/ZigbeeNet #111: Bug or wrong implementation?

LeifPetter, 2020-04-05:
Hi @Mr-Markus 
First, thank's for useful code :-)

I have one issue that stop me, and I can't find how to avoid this issue.

I'm using ConBee.

In **ZigBeeNode.cs**

        public ZigBeeEndpoint GetEndpoint(byte endpointId)
        {
            if (_endpoints.TryGetValue(endpointId, out ZigBeeEndpoint endpoint))
            {
                return endpoint;
            }

            return null;
        }

**_endpoint is always null._**

Is this an bug or is it me that have wrong implementation of the library?

LP :-)

[Program.zip](https://github.com/Mr-Markus/ZigbeeNet/files/4434189/Program.zip)

Mr-Markus, 2020-04-05:
Hi @LeifPetter ,

you are wellcome. I am happy that my code helps another people 😊

I took a look at your code and could not found any code that starts a discovery on your joined nodes. This will be necessary to get the node's "metadata" which contains the endpoints and it's supported clusters.

You need something like this:
```cs
ZigBeeDiscoveryExtension discoveryExtension = new ZigBeeDiscoveryExtension();
discoveryExtension.SetUpdatePeriod(60);
networkManager.AddExtension(discoveryExtension);
```

It's an integrated extension that adds a listener to the NetworkManager which starts node discovery when a new node is added to your network. Don't forget to use `NetworkManager.PermitJoin(byte duration)` for that.

Only if a node is discovered the GetEndpoint(byte endpointId) will return data

You also need to save the network state (joined nodes) to a database like Json, MongoDb etc. to get all joined devices and its "metadata" like endpoints after restarting your network. 
An example here: 
https://github.com/Mr-Markus/ZigbeeNet/blob/12e94f471cdc67e2f628f2a275be5e2bd8fa2f4d/samples/ZigBeeNet.PlayGround/Program.cs#L157-L161

For further information just comment below, create a new issue or see wiki [here](https://github.com/Mr-Markus/ZigbeeNet/wiki)

LeifPetter, 2020-04-06:
Hi @Mr-Markus 

Thanks for trying to help me 👍 
Sorry about the mess from me, but I want to use your library. I have 1x ConBee and 3x On/Off sensors that will be placed at doors.
I have tried to implement the code as I understood both from you and the examples you have, but still issue with endpoint that give me null.

Could you please have a new look at my PoC, and see what I'm doing wrong?

Thanks Markus 

LP

[HomeStatusService.zip](https://github.com/Mr-Markus/ZigbeeNet/files/4438056/HomeStatusService.zip)

Mr-Markus, 2020-04-07:
Hi @LeifPetter ,

up to line 54 everything is ok. But then you use some classes I don't know why they should be relevant for you. At first you should not need to call AddSupportedClusters, because this prevents messages from other clusters. Then you are using ZigBeeBasicServerExtension that does not make sense for me now.
After that you are allow joining devices for only 16 seconds. This should be at least 60 seconds, because some devices need more time to be detected after you activated it's join mode.

Which devices exactly would you use? I don't understand how you understand On/Off sensor in combination with doors? An On/Off device is an bulb or switch for example. If you mean a sensor that changes state from closed to open it's not On/Off.

And than before you startup the NetworkManager you try to update nodes with an empty node object for specified network and ieeeAddresses. This would clear all "metadata" if they will already exist.

I would recommend you to remove the code where you read from your config, start the application, activate join mode of your devices within 60 seconds and wait a few seconds until discovery is started. After discovery was sucessfully finished you should have endpoints.

I hope this will be helpful for you.

LeifPetter, 2020-04-08:
Hi @Mr-Marcus
Thank you for all your support and your try to help me 

I know you really try to help me, but I’m stuck.

In ZigBeeNode.cs I have added code like this:

        public ZigBeeEndpoint GetEndpoint(byte endpointId)
        {
            string info;

            if (_endpoints.TryGetValue(endpointId, out ZigBeeEndpoint endpoint))
            {
                info = "GetEndpoint - NOT Null !!";
                Log.Information(info);
                Console.WriteLine(info);

                return endpoint;
            }

            info = "GetEndpoint == Null";
            Log.Information(info);
            Console.WriteLine(info);
            return null;
        }

I only read from config file if there is no existing json storage files.
I have also tried to move “Startup” to different places in the code.

My ZigBee devices are magnetic ON/OFF Sensors. Should I add any extra code for this?

I also try to add ConBee USB plug as a node, just to see if there was any changes in behavior.

Sorry @Mr-Marcus, I really want to use your library. I'm not in details with ZigBee, this is my first project, so I can have missed to important information I should implement in my code.

I have also added logfile

Still null when reading endpoints
Any idea why it is still not working as expected?

[HomeStatusService.zip](https://github.com/Mr-Markus/ZigbeeNet/files/4450137/HomeStatusService.zip)
[HomeStatus.zip](https://github.com/Mr-Markus/ZigbeeNet/files/4450140/HomeStatus.zip)

Mr-Markus, 2020-04-08:
Hi @LeifPetter ,

you mentioned the wrong markus ;-)

No problem, everybody is starting with his first project in his life. It is no shame if you don't know how something works at the first time.

At first I would recommend you to create a own repository with your code. Adding as zip file in comments is hard to handle. With an own repository I can send you a PR with my code suggestions. 

Second I can tell you that you don't need to modify ZigBeeNet code, but only reference it in your project (see Playground app).

As third you don't need a settings file, because storage file will be created if new devices join your network. And joining is necessary because that will ask your devices for it's data like endpoints etc.
Only after discovery task was running after it joined your network you will be able to talk with your device.

LeifPetter, 2020-04-08:
Hi @Mr-Markus
Sorry for incorrect "Marcus"

Could see it could be misunderstood. I have been developer/head of development for 30 years. This is my first ZigBee project, not first project ;-)

Anyway, I will take a look later to check why it does not work as expected for me

LeifPetter, 2020-04-08:
Hi @Mr-Markus
Sorry for incorrect "Marcus"

Less misunderstanding. I have been developer/head of development for 30 years. This is my first ZigBee project, not first project ;-)

Anyway, I will take a look later to check why it does not work as expected for me

closed 2020-04-08 by LeifPetter