Mr-Markus/ZigbeeNet #117: Discovering nodes

genemars, 2020-06-17:
Hi, I have a *Philips Hue Bridge* with a few hue strips and lights.
They are all paired and working with the hue bridge.
How can I discover/add those nodes and control them with ZigbeeNet?
I tried the sample app but it's asking me a destination address which I have no clue where to get it.
Also I had to comment two lines of code because the sample app was throwing an exception:
```
                //Console.WriteLine($"Network Key      = {networkManager.ZigBeeNetworkKey}");
                //Console.WriteLine($"Link Key         = {networkManager.ZigBeeLinkKey}");
```
**Exception**
```
Channel          = CHANNEL_15
[13:12:04 DBG] Slip Send:0A-08-00-08-00-01-00-18-CD-FF
[13:12:04 DBG] Slip Recv:0A-08-07-08-00-01-00-18-C6-FF
System.AggregateException: One or more errors occurred. (INVALID_VALUE)
 ---> System.Exception: INVALID_VALUE
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at ZigbeeNet.Hardware.ConBee.ConBeeInterface.ReadParameter(Parameters parameter) in /home/gene/Scaricati/ZigbeeNet-develop/libraries/ZigbeeNet.Hardware.ConBee/Internal/ConBeeInterface.cs:line 217
   at ZigbeeNet.Hardware.ConBee.ConBeeInterface.get_NetworkKey() in /home/gene/Scaricati/ZigbeeNet-develop/libraries/ZigbeeNet.Hardware.ConBee/Internal/ConBeeInterface.cs:line 329
   at ZigbeeNet.Hardware.ConBee.ZigbeeDongleConBee.get_ZigBeeNetworkKey() in /home/gene/Scaricati/ZigbeeNet-develop/libraries/ZigbeeNet.Hardware.ConBee/ZigbeeDongleConBee.cs:line 258
   at ZigBeeNet.ZigBeeNetworkManager.get_ZigBeeNetworkKey() in /home/gene/Scaricati/ZigbeeNet-develop/libraries/ZigBeeNet/ZigBeeNetworkManager.cs:line 419
   at ZigBeeNet.PlayGround.Program.Main(String[] args) in /home/gene/Scaricati/ZigbeeNet-develop/samples/ZigBeeNet.PlayGround/Program.cs:line 204
```
I am using *Ubuntu 20* and a *ConBee II* dongle.

Mr-Markus, 2020-06-17:
Hi @genemars 
In the ZigBee specification is declared that a device can only be in one network at the same time. So you are Not able to control hue connected devices with this library. 
This library allows you to control devices without hue within an own network. Your dongle will take the same role as your Hje bridge.

If you would do this i suggest you to remove the bulb from hue network and start join mode with the sample app to attach the device to the New network. Then you will get a network address that you can use to on of off the device

Is this helpful?

genemars, 2020-06-17:
>If you would do this i suggest you to remove the bulb from hue network and start join mode with the sample app to attach the device to the New network. Then you will get a network address that you can use to on of off the device

Yes I already tried that, but after I type the `join` command, nothing happens.

```
cmd> join
[13:58:25 DBG] TX CMD: ManagementPermitJoiningRequest [0/0 -> 0/0, cluster=54, transId=5, PermitDuration=255, TcSignificance=True]
[13:58:25 DBG] TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=54, addressMode=Device, radius=31, apsCounter=5, payload=5 255 1]
[13:58:25 DBG] Slip Send:12-13-00-1F-00-18-00-05-00-03-A0-97-05-FF-FF-2E-21-00-00-00-00-36-00-00-03-00-05-FF-01-04-1F-B2-FA
[13:58:25 DBG] Slip Recv:12-13-00-09-00-02-00-22-05-A9-FF
[13:58:25 DBG] Slip Recv:0E-14-00-07-00-AE-00-29-FF
[13:58:25 DBG] Slip Send:17-14-00-07-00-00-00-CE-FF
[13:58:25 DBG] Slip Recv:17-14-00-20-00-19-00-26-02-00-00-00-02-00-00-00-00-00-36-80-02-00-05-00-00-AF-F1-8C-02-00-00-13-74-FC
[13:58:25 DBG] RX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=32822, addressMode=0, radius=0, apsCounter=0, payload=5 0]
[13:58:25 DBG] RX CMD: ManagementPermitJoiningResponse [0/0 -> 0/0, cluster=32822, transId=5, Status=SUCCESS]
[13:58:25 DBG] Slip Send:04-15-00-07-00-00-00-E0-FF
[13:58:25 DBG] Slip Recv:04-15-00-13-00-0C-00-22-05-02-00-00-00-00-00-00-00-00-00-9F-FF

1 node(s)

0. COORDINATOR: 0

cmd> 
```
Am I supposed to issue some other command then?
Shouldn't I get a list of nodes available for pairing or something like that?

Mr-Markus, 2020-06-17:
You are correct to execute the join command. But you have to get the device into join Mode. Which device do you have? 
You should find steps for your devices in the Web, too. This is different for any manufacturer and sometimes little bit stupid... Osram devices should be turned on and off in 5 sec intervals and 5 Times for example. For Hue bulbs you need a switch near to it

After your device is in join Mode the playground App would display it in console, because it waits for incomming messages

genemars, 2020-06-17:
I have a philips hue bulbs. They don't have any button. But the hue gateway can discover and add nodes without taking any action on them.
I will try switching the bulb on and off anyway.

genemars, 2020-06-17:
@Mr-Markus I've tried switching the bulb on and off with no luck with pairing.
Do I have to change the channel to 11 on the ConBee II which actually reports as 15? Is that relevant?

Mr-Markus, 2020-06-17:
Try this https://github.com/mozilla-iot/wiki/wiki/HOWTO:-Factory-reset-a-Hue-bulb

genemars, 2020-06-17:
It didn't work. Anyway it seems I can't make this work with the ConBee II software neither (deCONZ, just tried it).

Mr-Markus, 2020-06-17:
It should work. Maybe @DavidKarlas could help you? He implememted conbee support

Mr-Markus, 2020-06-17:
Did your bulb Flash after Set it to join Mode with your switch? If there are no Errors in the console no packages are received.

genemars, 2020-06-18:
It didn't flash.
It looks like the only solution is this one:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1878#issuecomment-533782102

Mr-Markus, 2020-06-18:
Yes, that is what I meant when I said you have to get it to join mode, but every manufacturer implements it another way. And Philips uses the I/O switch. You have to hold it next to the bulb (a few centimeter) and push I and O Button and hold it a few seconds until the bulb will flash.
Then the bulb is in join mode and the console app will display a few messages, because it is discovering the clusters and attributes and add the device to a local storage to save your network state (devices in your network).

After that you are able to send a command to a network address. The network address is unique within a network (don't mix up with IeeeAddress, this is a kind of a MAC address for devices). Everytime you reset a device and rejoin it to a network the network address will change, but IeeeAddress will always be the same. For sending commands you need the network address.

It is possible to get the network address by IeeeAddress and it is sending a broadcast to the network and waits for an response of the given device, but this is more advanced.
For your case you should just need to use the I/O switch to get it into join mode.

Feel free to use the playground app or build your own with ZigBeeNet

genemars, 2020-06-18:
@Mr-Markus I ordered the I/O switch and I will be able to test node reset as soon as I get it.
I will use the playground app to familiarize with ZigBeeNet, after that my aim is to integrate it in HomeGenie.
Thanks for you kind support  =)

Mr-Markus, 2020-06-18:
You're welcome 😊👍

genemars, 2020-06-19:
Hi @Mr-Markus I just received the philips hue remote and I was able to reset and make a bulb pair with the conbee.
But now ZigbeeNet is unable to determine the network address of the device I guess:
```
[12:49:23 DBG] Transaction timeout: NetworkAddressRequest [0/0 -> 65535/0, cluster=0, transId=3, IeeeAddr=0017880100B1ADEF, RequestType=0, StartIndex=0]
[12:49:23 DBG] 0017880100B1ADEF: Node SVC Discovery: NetworkAddressRequest returned null
[12:49:23 DBG] 0017880100B1ADEF: Node SVC Discovery: request NWK_ADDRESS failed. Retry 1, wait 1 ms before retry.
[12:49:23 DBG] 0017880100B1ADEF: Node SVC Discovery: running
[12:49:23 DBG] TX CMD: NetworkAddressRequest [0/0 -> 65535/0, cluster=0, transId=6, IeeeAddr=0017880100B1ADEF, RequestType=0, StartIndex=0]
[12:49:23 DBG] TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65535/0, profile=0, cluster=0, addressMode=Device, radius=31, apsCounter=6, payload=6 239 173 177 0 1 136 23 0 0 0]
[12:49:23 DBG] Slip Send:12-15-00-21-00-1A-00-06-00-02-FF-FF-00-00-00-00-00-00-0B-00-06-EF-AD-B1-00-01-88-17-00-00-00-04-1F-77-FA
[12:49:23 DBG] Slip Recv:12-15-00-09-00-02-00-22-06-A6-FF
[12:49:23 DBG] Slip Recv:0E-16-00-07-00-A6-00-2F-FF
[12:49:23 DBG] Slip Send:04-16-00-07-00-00-00-DF-FF
[12:49:23 DBG] Slip Recv:04-16-00-13-00-0C-00-22-06-02-FF-FF-00-00-A6-00-00-00-00-F9-FC
```
and this is the list of nodes:
```
2 node(s)

0. UNKNOWN: 56715
1. COORDINATOR: 0
```
And this is what happens when I try a `off` command:
```
cmd> off 
Destination Address: 56715
No endpoint found
```
What am I doing wrong here? Maybe I picked the wrong number as network address?

genemars, 2020-06-19:
[image]

It looks like the node is found, but it has no `endpoint` (not sure of what this means).

genemars, 2020-06-19:
**Additional details:** if I turn off and on the bulb this is what I get in the console:
```

cmd> [14:49:12 DBG] Slip Recv:0E-30-00-07-00-AA-00-11-FF
[14:49:12 DBG] Slip Send:17-30-00-07-00-00-00-B2-FF
[14:49:12 DBG] Slip Recv:17-30-00-2A-00-23-00-22-02-FD-FF-00-02-8B-DD-00-00-00-13-00-0C-00-00-8B-DD-EF-AD-B1-00-01-88-17-00-8E-00-AF-FF-05-22-00-00-CF-3C-F4
[14:49:12 DBG] RX APS: ZigBeeApsFrame [sourceAddress=56715/0, destinationAddress=65533/0, profile=0, cluster=19, addressMode=0, radius=0, apsCounter=0, payload=0 139 221 239 173 177 0 1 136 23 0 142]
[14:49:12 DBG] RX CMD: DeviceAnnounce [56715/0 -> 65533/0, cluster=19, transId=0, NwkAddrOfInterest=56715, IeeeAddr=0017880100B1ADEF, Capability=142]
[14:49:12 DBG] DISCOVERY Extension: Mesh related command received. Triggering mesh update.
[14:49:12 DBG] DISCOVERY Extension: Start mesh update task with interval of 60 seconds
[14:49:12 DBG] 0017880100B1ADEF: Device announce received. NWK=56715
[14:49:12 DBG] 0017880100B1ADEF: NWK Discovery add node 56715
[14:49:12 DBG] 0017880100B1ADEF: Node state updated from UNKNOWN to ONLINE
[14:49:12 DBG] 0017880100B1ADEF: Updating Node 56715
[14:49:12 DBG] 0017880100B1ADEF: Node 56715 update
[14:49:12 DBG] 0017880100B1ADEF: Node 56715 is not updated
[14:49:12 DBG] DISCOVERY Extension: Starting mesh update
[14:49:12 DBG] 0017880100B1ADEF: DISCOVERY Extension: Starting mesh update
[14:49:12 DBG] 0017880100B1ADEF: Node SVC Discovery: Update mesh not performed for possible end device
[14:49:12 DBG] 00212EFFFF0597A0: DISCOVERY Extension: Starting mesh update
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: Update mesh
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: starting new tasks ["NEIGHBORS", "ROUTES"]
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: scheduled ["NEIGHBORS", "ROUTES"]
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: running
[14:49:12 DBG] TX CMD: ManagementLqiRequest [0/0 -> 0/0, cluster=49, transId=17, StartIndex=0]
[14:49:12 DBG] TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=49, addressMode=Device, radius=31, apsCounter=17, payload=17 0]
[14:49:12 DBG] Slip Send:12-31-00-1E-00-17-00-11-00-03-A0-97-05-FF-FF-2E-21-00-00-00-00-31-00-00-02-00-11-00-04-1F-84-FB
[14:49:12 DBG] Slip Recv:12-31-00-09-00-02-00-22-11-7F-FF
[14:49:12 DBG] Slip Recv:0E-32-00-07-00-AE-00-0B-FF
[14:49:12 DBG] Slip Send:17-32-00-07-00-00-00-B0-FF
[14:49:12 DBG] Slip Recv:17-32-00-39-00-32-00-26-02-00-00-00-02-00-00-00-00-00-31-80-1B-00-11-00-01-00-01-A0-97-05-FF-FF-2E-21-00-EF-AD-B1-00-01-88-17-00-8B-DD-25-01-01-FF-00-AF-00-00-00-00-00-00-90-F4
[14:49:12 DBG] RX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=32817, addressMode=0, radius=0, apsCounter=0, payload=17 0 1 0 1 160 151 5 255 255 46 33 0 239 173 177 0 1 136 23 0 139 221 37 1 1 255]
[14:49:12 DBG] RX CMD: ManagementLqiResponse [0/0 -> 0/0, cluster=32817, transId=17, Status=SUCCESS, NeighborTableEntries=1, StartIndex=0, NeighborTableList=NeighborTable [extendedPanId=00212EFFFF0597A0, extendedAddress=0017880100B1ADEF, networkAddress=56715, deviceType=ROUTER, rxOnWhenIdle=RX_ON, relationship=SIBLING, permitJoining=ENABLED, depth=1, lqi=255]]
[14:49:12 DBG] Transaction complete: ManagementLqiRequest [0/0 -> 0/0, cluster=49, transId=17, StartIndex=0]
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: ManagementLqiRequest response CommandResult [SUCCESS, ManagementLqiResponse [0/0 -> 0/0, cluster=32817, transId=17, Status=SUCCESS, NeighborTableEntries=1, StartIndex=0, NeighborTableList=NeighborTable [extendedPanId=00212EFFFF0597A0, extendedAddress=0017880100B1ADEF, networkAddress=56715, deviceType=ROUTER, rxOnWhenIdle=RX_ON, relationship=SIBLING, permitJoining=ENABLED, depth=1, lqi=255]]]
[14:49:12 DBG] Slip Send:04-33-00-07-00-00-00-C2-FF
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: ManagementLqiRequest complete [1 neighbors]
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: request NEIGHBORS successful.
[14:49:12 DBG] Slip Recv:04-33-00-13-00-0C-00-22-11-02-00-00-00-00-00-00-00-00-00-75-FF
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: running
[14:49:12 DBG] TX CMD: ManagementRoutingRequest [0/0 -> 0/0, cluster=50, transId=18, StartIndex=0]
[14:49:12 DBG] TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=50, addressMode=Device, radius=31, apsCounter=18, payload=18 0]
[14:49:12 DBG] Slip Send:12-34-00-1E-00-17-00-12-00-03-A0-97-05-FF-FF-2E-21-00-00-00-00-32-00-00-02-00-12-00-04-1F-7E-FB
[14:49:12 DBG] Slip Recv:12-34-00-09-00-02-00-22-12-7B-FF
[14:49:12 DBG] Slip Recv:0E-35-00-07-00-AE-00-08-FF
[14:49:12 DBG] Slip Send:17-35-00-07-00-00-00-AD-FF
[14:49:12 DBG] Slip Recv:17-35-00-23-00-1C-00-26-02-00-00-00-02-00-00-00-00-00-32-80-05-00-12-00-00-00-00-00-AF-11-06-22-00-00-15-85-FD
[14:49:12 DBG] RX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=32818, addressMode=0, radius=0, apsCounter=0, payload=18 0 0 0 0]
[14:49:12 DBG] RX CMD: ManagementRoutingResponse [0/0 -> 0/0, cluster=32818, transId=18, Status=SUCCESS, RoutingTableEntries=0, StartIndex=0, RoutingTableList=]
[14:49:12 DBG] Transaction complete: ManagementRoutingRequest [0/0 -> 0/0, cluster=50, transId=18, StartIndex=0]
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: ManagementRoutingRequest returned CommandResult [SUCCESS, ManagementRoutingResponse [0/0 -> 0/0, cluster=32818, transId=18, Status=SUCCESS, RoutingTableEntries=0, S [... cut ...]
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: ManagementLqiRequest complete [0 routes]
[14:49:12 DBG] Slip Send:04-36-00-07-00-00-00-BF-FF
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: request ROUTES successful.
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: running
[14:49:12 DBG] 00212EFFFF0597A0: Node SVC Discovery: complete
[14:49:12 DBG] 00212EFFFF0597A0: Updating Node 0
[14:49:12 DBG] 00212EFFFF0597A0: Node 0 update
[14:49:12 DBG] Slip Recv:04-36-00-13-00-0C-00-22-12-02-00-00-00-00-00-00-00-00-00-71-FF
[14:49:12 DBG] 00212EFFFF0597A0: Node 0 is not updated
```

nicolaiw, 2020-06-19:
Hi,

after that (on->off), what does the cmd "endpoints" say?

Best regards

genemars, 2020-06-19:
When I turn the bulb off and then on I can see a series of failed requests:
```
2 node(s)

0. UNKNOWN: 1627
1. COORDINATOR: 0

cmd> [16:02:29 DBG] Transaction timeout: NetworkAddressRequest [0/0 -> 65535/0, cluster=0, transId=15, IeeeAddr=0017880100B1ADEF, RequestType=0, StartIndex=0]
[16:02:29 DBG] 0017880100B1ADEF: Node SVC Discovery: NetworkAddressRequest returned null
[16:02:29 DBG] 0017880100B1ADEF: Node SVC Discovery: request NWK_ADDRESS failed after 9 attempts.
[16:02:29 DBG] 0017880100B1ADEF: Node state updated from UNKNOWN to OFFLINE
[16:02:29 DBG] 0017880100B1ADEF: Node SVC Discovery: stopped
```
and if I issue the `endpoints` command:
```
cmd> endpoints
Destination Address: 1627
--------------------

--------------------

2 node(s)

0. UNKNOWN: 1627
1. COORDINATOR: 0
```

nicolaiw, 2020-06-19:
mhm .. I guess this Issue will be hard to discover without Debugging with the same Dongle and Bulb. @Mr-Markus any Ideas? :D

genemars, 2020-06-19:
This is the log of a *NetworkAddressRequest* transaction:
```
[16:35:19 DBG] 0017880100B1ADEF: Node SVC Discovery: running
[16:35:19 DBG] TX CMD: NetworkAddressRequest [0/0 -> 65535/0, cluster=0, transId=13, IeeeAddr=0017880100B1ADEF, RequestType=0, StartIndex=0]
[16:35:19 DBG] TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65535/0, profile=0, cluster=0, addressMode=Device, radius=31, apsCounter=13, payload=13 239 173 177 0 1 136 23 0 0 0]
[16:35:19 DBG] Slip Send:12-25-00-21-00-1A-00-0D-00-02-FF-FF-00-00-00-00-00-00-0B-00-0D-EF-AD-B1-00-01-88-17-00-00-00-04-1F-59-FA
[16:35:19 DBG] Slip Recv:12-25-00-09-00-02-00-22-0D-8F-FF
[16:35:19 DBG] Slip Recv:0E-26-00-07-00-A6-00-1F-FF
[16:35:19 DBG] Slip Send:04-26-00-07-00-00-00-CF-FF
[16:35:19 DBG] Slip Recv:04-26-00-13-00-0C-00-22-0D-02-FF-FF-00-00-A6-00-00-00-00-E2-FC
[16:35:27 DBG] Transaction timeout: NetworkAddressRequest [0/0 -> 65535/0, cluster=0, transId=13, IeeeAddr=0017880100B1ADEF, RequestType=0, StartIndex=0]
[16:35:27 DBG] 0017880100B1ADEF: Node SVC Discovery: NetworkAddressRequest returned null
[16:35:27 DBG] 0017880100B1ADEF: Node SVC Discovery: request NWK_ADDRESS failed after 9 attempts.
[16:35:27 DBG] 0017880100B1ADEF: Node state updated from UNKNOWN to OFFLINE
[16:35:27 DBG] 0017880100B1ADEF: Node SVC Discovery: stopped
```
The bulb replies something, but it seems it's not the expected input.

genemars, 2020-06-19:
And this is the bulb node info

[image]

spudwebb, 2020-06-19:
The problem probably comes from the Conbee driver, I don't think it has been tested by a lot of people.

My Philips hue bulbs work fine with both Digi and Ember dongles

Mr-Markus, 2020-06-19:
It seems that endpoints were not discovered correctly. I had this with osram bulbs an CC2531 too. I thought it Was a problem with the osram bulb, but Not for Sure.
If you restart the App and turn the bulb of and on it works (sometimes)
But this does not solve the problem...

Any one have an idea or having this problem too?

genemars, 2020-06-22:
I just received a new device, which is **Innr** *smart filament bulb*.
It behaves the same way as described for Philips Hue bulb.
Can someone show me the log of a correct `NetworkAddressRequest` transaction?
Maybe I can spot what's wrong looking at the differences in the node reply.

genemars, 2020-06-22:
@Mr-Markus I just managed to make discovery works by commenting one line of code:

https://github.com/Mr-Markus/ZigbeeNet/blob/8c1ff74362aa383805440862661632385dd766bd/libraries/ZigBeeNet/App/Discovery/ZigBeeNodeServiceDiscoverer.cs#L247

if I comment this line everything works, the endpoints are discovered and I can control both the Philips Hue and the Innr bulbs.
So it looks like that something needs to be fixed around that line.
Maybe some devices with old Zigbee stack version don't understand that kind of request?

Mr-Markus, 2020-06-22:
I think about if this step is really necessary, because the network address will not change for an device within a network. It is just for sure that no invalid devices are in the network state.
If a device will leave it is removed and if added it is initially discovered and state is saved.

That would be possible that older ZigBee stack version would not understand it, but I am not sure.

So it should be ok to skip this step. You just have to check that while adding the device via join mode all endpoints and other requests were done correctly.

Let me know if this worked for you

genemars, 2020-06-22:
All other discovery tasks were completed successfully.
I think it's correct to assume that once joined the network, a device will not change its network address.
Let me know if there is something else specific you want me to test.

closed 2020-08-01 by Mr-Markus

reinux, 2020-09-11:
I'm getting this problem with an Ikea Tradfri E26 and ConBee II.

I also see no endpoints when I set up a simple program instead of using the PlayGround.

Cypher87, 2020-12-23:
I had the same issue. Changing the line @genemars mentioned to `success = true;` fixed it for me (using TiCC2531)