Mr-Markus/ZigbeeNet #118: Serial Port errors

genemars, 2020-06-22:
Occasionally I get:
```
[17:03:35 ERR] Error while reading from serial port: The port is closed.
System.ObjectDisposedException: The port is closed.
   at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
   at System.IO.Ports.SerialStream.Read(Byte[] array, Int32 offset, Int32 count)
   at System.IO.Ports.SerialPort.Read(Byte[] buffer, Int32 offset, Int32 count)
   at ZigBeeNet.Tranport.SerialPort.ZigBeeSerialPort.ReaderTask() in /home/gene/Scaricati/ZigbeeNet-develop/libraries/ZigBeeNet.Transport.SerialPort/ZigBeeSerialPort.cs:line 185
```
My configuration is Ubuntu + ConBee II and mono version 6.8.0.
Is there any way (event?) to detect this error and make the serialport connect again?

Mr-Markus, 2020-06-22:
Hi @genemars ,

When do you receive this error? During startup or while running the app?

Because this libary is using netstandard2.0 and Playground app is running with netcore3.0 you don't need mono. With .Net Core 3.0 SerialPort is fully supported.
Did you compile the Playground app under Linux? I've tested it under Ubuntu 18.04 and it worked.

Your goal should be to get it work this way.

genemars, 2020-06-22:
Yes, right, it's not running via `mono` but via `dotnet v3.1.301` (Ubuntu 20.04 LTS).
Everything compiles and runs nicely, but while running, it seems to me during idle times, the sample app throws that error.

Mr-Markus, 2020-06-22:
I am sure that is not a bug in ZigBeeNet, but in Linux or dotnet implementation under linux. ZigBeeNet is only closing it wby calling NetworkManager.Shutdown() or by Exception. Any outputs?

Because it is Ubuntu 20.04 which is just a few weeks old it is possible that there are some changes, that is not fixed in latest net core version.
We should look here: https://github.com/dotnet/runtime/issues?q=+serialport+

genemars, 2020-06-24:
Yeah, .Net SerialPort implementation is not that good and it always had issues in `Linux` since ever.
That's why I had to implement my own `SerialPortLib` to use it in my other projects such as `ZWaveLib`.
So this is to be considered a minor issue not related to `ZigbeeNet`.

Mr-Markus, 2020-06-24:
You can implement your SerialPortLib with the `IZigBeePort` interface like here
https://github.com/Mr-Markus/ZigbeeNet/blob/develop/libraries/ZigBeeNet.Transport.SerialPort/ZigBeeSerialPort.cs

Mr-Markus, 2020-06-30:
I close this issue, because it is not a bug of ZigBeeNet

closed 2020-06-30 by Mr-Markus