Mr-Markus/ZigbeeNet #120: Tested device - CC2531 - Aqara WSDCGQ11LM Temperature Humidity Sensor (Xiaomi)

dobrou, 2020-07-28:
Hi,
I'm reporting a working device.

Using code snippets from playground example,
using usb stick CC2531,
I successfully received temperature/pressure/humidity from Aqara WSDCGQ11LM Temperature Humidity Sensor ( Xiaomi Ecosysterm Product )

Thank you for this library, nice work! :)

Mr-Markus, 2020-08-01:
Hi @dobrou ,

thanks for your report. I added it to our wiki page here https://github.com/Mr-Markus/ZigbeeNet/wiki/Supported-devices

I am happy that you use and like ZigBeeNet 😊

Best
Markus

closed 2020-08-01 by Mr-Markus

jgmdavies, 2020-08-07:
Hi @dobrou,

Many thanks.  I've got one of these - does it definitely need the 'Xiaomi home app', as mentioned by some reviewers?

I'm using a ConBee II and DeConz at the moment, but can't see the sensor.

Thanks,
Jim

Mr-Markus, 2020-08-07:
Hi @jgmdavies ,

where did you found this information? I haven't heard of this before... I use this sensors, too but i didn't need an app for it. ZigBee devices should work independently with ZigBeeNet. You just need to get it into learn mode. This should be possible with this sensor by pressing the button a few seconds after you started join mode in your app with ZigBeeNet

Best
Markus

TheBlubb14, 2020-08-07:
> 
> 
> Hi @dobrou,
> 
> Many thanks. I've got one of these - does it definitely need the 'Xiaomi home app', as mentioned by some reviewers?
> 
> I'm using a ConBee II and DeConz at the moment, but can't see the sensor.
> 
> Thanks,
> Jim

No you dont need "Xiaomi home app". I have multiple connected with my CC2531 and i dont use "Xiaomi home app".

jgmdavies, 2020-08-07:
Many thanks @Mr-Markus and @TheBlubb14, I'll have another go.

The app was mentioned in some Amazon customer reviews, e.g. (translated from Deutsch):

"_Not to use without Aqara Hub. This can only be understood at the third glance from the miserable article description due to lack of translation._"

lol

jgmdavies, 2020-08-07:
Sorry folks, my fault.  I didn't realise that I had to 'Add new sensor' in Phoscon - I assumed that deCONZ would see it.

It's working fine now, so that's an Aqara WSDCGQ11LM + ConBee II + deCONZ  + Phoscon App, if anyone's interested.

I'll try the ZigbeeNet Playground next.

jgmdavies, 2020-08-07:
Should the Playground sample just see the Aqara sensor now, without needing any commands?

It only seems to see the ConBee:

```
Network state is updated to INITIALISING: Emit
"00212EFFFF042B9F": Data store: Node was restored.: Emit
"00212EFFFF042B9F": Updating Node 0: Emit
Opening port "COM5" at 9600 baud with FLOWCONTROL_OUT_NONE: Emit
Slip Send:0D-01-00-05-00-ED-FF: Emit
Slip Recv:0D-01-00-09-00-00-07-58-26-64-FF: Emit
Slip Send:0A-02-00-08-00-01-00-01-EA-FF: Emit
Slip Recv:0A-02-00-10-00-09-00-01-9F-2B-04-FF-FF-2E-21-00-BF-FC: Emit
Slip Send:0A-03-00-08-00-01-00-07-E3-FF: Emit
Slip Recv:0A-03-00-0A-00-03-00-07-00-00-DF-FF: Emit
Slip Send:07-04-00-08-00-00-00-00-ED-FF: Emit
Slip Recv:07-04-00-07-00-A2-00-4C-FF: Emit
NetworkStateChanged:NET_CONNECTED: Emit
Adding supported cluster 6: Emit
ClusterMatcher starting: Emit
ClusterMatcher adding cluster 6: Emit
Adding supported cluster 768: Emit
ClusterMatcher adding cluster 768: Emit
Network state: INITIALISING
Slip Send:0A-05-00-08-00-01-00-05-E3-FF: Emit
Slip Recv:0A-05-00-0A-00-03-00-05-0B-84-50-FF: Emit
PAN ID           = 33803
Slip Send:0A-06-00-08-00-01-00-08-DF-FF: Emit
Slip Recv:0A-06-00-10-00-09-00-08-9F-2B-04-FF-FF-2E-21-00-B4-FC: Emit
Extended PAN ID  = 00212EFFFF042B9F
Slip Send:0A-07-00-08-00-01-00-1C-CA-FF: Emit
Slip Recv:0A-07-00-09-00-02-00-1C-0F-B9-FF: Emit
Channel          = CHANNEL_15
ZigBeeNetworkManager startup: reinitialize=False, networkState=INITIALISING: Emit
Network state is updated to ONLINE: Emit
ZigBee console starting up ... [OK]: Emit
Permit join to "65532/0" for 0 seconds.: Emit
TX CMD: "ManagementPermitJoiningRequest [0/0 -> 65532/0, cluster=54, transId=1, PermitDuration=0, TcSignificance=True]": Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65532/0, profile=0, cluster=54, addressMode=Device, radius=31, apsCounter=1, payload=1 0 1]": Emit
Slip Send:12-08-00-19-00-12-00-01-00-02-FC-FF-00-00-00-36-00-00-03-00-01-00-01-04-1F-5F-FD: Emit
Slip Recv:12-08-00-09-00-02-00-22-01-B8-FF: Emit
Slip Recv:0E-09-00-07-00-A6-00-3C-FF: Emit
TX CMD: "ManagementPermitJoiningRequest [0/0 -> 0/0, cluster=54, transId=2, PermitDuration=0, TcSignificance=True]": Emit
Slip Send:04-09-00-07-00-00-00-EC-FF: Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=54, addressMode=Device, radius=31, apsCounter=2, payload=2 0 1]": Emit
Slip Recv:04-09-00-13-00-0C-00-22-01-02-FC-FF-00-00-A6-00-00-00-00-0E-FD: Emit
DISCOVERY Extension: Startup: Emit
Slip Send:12-0A-00-1F-00-18-00-02-00-03-9F-2B-04-FF-FF-2E-21-00-00-00-00-36-00-00-03-00-02-00-01-04-1F-2E-FC: Emit
Slip Recv:12-0A-00-09-00-02-00-22-02-B5-FF: Emit
Slip Recv:0E-0B-00-07-00-AE-00-32-FF: Emit
Network discovery task: starting: Emit
Slip Send:17-0B-00-07-00-00-00-D7-FF: Emit
DISCOVERY Extension: Starting mesh update: Emit
"00212EFFFF042B9F": DISCOVERY Extension: Adding discoverer for added node: Emit
Node 00212EFFFF042B9F added ZigBeeNode [IEEE=00212EFFFF042B9F, NWK=0x0000, Type=UNKNOWN]
Slip Recv:17-0B-00-20-00-19-00-26-02-00-00-00-02-00-00-00-00-00-36-80-02-00-02-00-00-AF-11-41-67-00-00-15-44-FD: Emit
RX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=32822, addressMode=0, radius=0, apsCounter=0, payload=2 0]": Emit
"00212EFFFF042B9F": Data store: Deferring write for 250ms.: Emit
"00212EFFFF042B9F": Node SVC Discovery: start discovery: Emit
"00212EFFFF042B9F": Node SVC Discovery: starting new tasks [NODE_DESCRIPTOR, POWER_DESCRIPTOR]: Emit
"00212EFFFF042B9F": Node SVC Discovery: scheduled [NODE_DESCRIPTOR, POWER_DESCRIPTOR]: Emit
RX CMD: "ManagementPermitJoiningResponse [0/0 -> 0/0, cluster=32822, transId=2, Status=SUCCESS]": Emit
Slip Send:04-0C-00-07-00-00-00-E9-FF: Emit
Slip Recv:04-0C-00-13-00-0C-00-22-02-02-00-00-00-00-00-00-00-00-00-AB-FF: Emit
"00212EFFFF042B9F": Node SVC Discovery: running: Emit
TX CMD: "NodeDescriptorRequest [0/0 -> 0/0, cluster=2, transId=3, NwkAddrOfInterest=0]": Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=2, addressMode=Device, radius=31, apsCounter=3, payload=3 0 0]": Emit
Slip Send:12-0D-00-1F-00-18-00-03-00-03-9F-2B-04-FF-FF-2E-21-00-00-00-00-02-00-00-03-00-03-00-00-04-1F-5E-FC: Emit
Slip Recv:12-0D-00-09-00-02-00-22-03-B1-FF: Emit
Slip Recv:0E-0E-00-07-00-AE-00-2F-FF: Emit
[... 33 lines cut ...]
Network state: ONLINE
Slip Send:0A-13-00-08-00-01-00-05-D5-FF: Emit
Slip Recv:0A-13-00-0A-00-03-00-05-0B-84-42-FF: Emit
PAN ID           = 33803
Slip Send:0A-14-00-08-00-01-00-08-D1-FF: Emit
Slip Recv:0A-14-00-10-00-09-00-08-9F-2B-04-FF-FF-2E-21-00-A6-FC: Emit
Extended PAN ID  = 00212EFFFF042B9F
Slip Send:0A-15-00-08-00-01-00-1C-BC-FF: Emit
Slip Recv:0A-15-00-09-00-02-00-1C-0F-AB-FF: Emit
Channel          = CHANNEL_15
Joining enabled...

1 node(s)

0. COORDINATOR: 0

1 node(s)

0. COORDINATOR: 0

1 node(s)

0. COORDINATOR: 0

```

jgmdavies, 2020-08-07:
If I do a quick press on the sensor, I get the below, which includes:

`Emit Incoming message from unknown node "AEA2": Notifying announce listeners`

but still only one node listed:

`1 node(s)
0. COORDINATOR: 0`

Sorry about the mysterious lack of line breaks:

`Slip Recv:0E-16-00-07-00-AA-00-2B-FF: Emit
Slip Send:17-16-00-07-00-00-00-CC-FF: Emit
Slip Recv:0E-17-00-07-00-AA-00-2A-FF: Emit
Slip Recv:0E-18-00-07-00-AA-00-29-FF: Emit
Slip Recv:17-16-00-26-00-1F-00-2A-02-00-00-01-02-A2-AE-01-04-01-02-04-08-00-18-2F-0A-00-00-29-F3-0A-00-AF-FF-4A-67-00-00-F6-2F-F9: Emit
RX APS: "ZigBeeApsFrame [sourceAddress=44706/1, destinationAddress=0/1, profile=260, cluster=1026, addressMode=0, radius=0, apsCounter=0, payload=24 47 10 0 0 41 243 10]": Emit
Incoming message from unknown node "AEA2": Notifying announce listeners: Emit
44706: NWK Discovery scheduling node discovery: Emit
44706: NWK Discovery starting node discovery: Emit
RX ZCL: "ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=False, direction=SERVER_TO_CLIENT, disableDefaultResponse=True, manufacturerCode=0, sequenceNumber=47, commandId=10]": Emit
Unknown node 44706: Emit
Incoming message did not translate to command.: Emit
Slip Send:17-17-00-07-00-00-00-CB-FF: Emit
Slip Recv:17-17-00-26-00-1F-00-2A-02-00-00-01-02-A2-AE-01-04-01-05-04-08-00-18-30-0A-00-00-21-60-11-00-AF-FF-4A-67-00-00-F6-BE-F9: Emit
RX APS: "ZigBeeApsFrame [sourceAddress=44706/1, destinationAddress=0/1, profile=260, cluster=1029, addressMode=0, radius=0, apsCounter=0, payload=24 48 10 0 0 33 96 17]": Emit
TX CMD: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=5, NwkAddrOfInterest=44706, RequestType=0, StartIndex=0]": Emit
Incoming message from unknown node "AEA2": Notifying announce listeners: Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0, cluster=1, addressMode=Device, radius=31, apsCounter=5, payload=5 162 174 0 0]": Emit
44706: NWK Discovery node discovery already in progress: Emit
RX ZCL: "ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=False, direction=SERVER_TO_CLIENT, disableDefaultResponse=True, manufacturerCode=0, sequenceNumber=48, commandId=10]": Emit
Unknown node 44706: Emit
Incoming message did not translate to command.: Emit
Slip Send:17-18-00-07-00-00-00-CA-FF: Emit
Slip Recv:17-18-00-2F-00-28-00-22-02-00-00-01-02-A2-AE-01-04-01-03-04-11-00-18-31-0A-00-00-29-F5-03-14-00-28-FF-10-00-29-96-27-00-AF-FF-4A-67-00-00-F5-EC-F6: Emit
RX APS: "ZigBeeApsFrame [sourceAddress=44706/1, destinationAddress=0/1, profile=260, cluster=1027, addressMode=0, radius=0, apsCounter=0, payload=24 49 10 0 0 41 245 3 20 0 40 255 16 0 41 150 39]": Emit
Incoming message from unknown node "AEA2": Notifying announce listeners: Emit
44706: NWK Discovery node discovery already in progress: Emit
RX ZCL: "ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=False, direction=SERVER_TO_CLIENT, disableDefaultResponse=True, manufacturerCode=0, sequenceNumber=49, commandId=10]": Emit
Unknown node 44706: Emit
Incoming message did not translate to command.: Emit
Slip Send:12-19-00-1B-00-14-00-05-00-02-FD-FF-00-00-00-01-00-00-05-00-05-A2-AE-00-00-04-1F-25-FC: Emit
Slip Recv:12-19-00-09-00-02-00-22-05-A3-FF: Emit
Slip Recv:0E-1A-00-07-00-A6-00-2B-FF: Emit
Slip Send:04-1A-00-07-00-00-00-DB-FF: Emit
Slip Recv:04-1A-00-13-00-0C-00-22-05-02-FD-FF-00-00-A6-00-00-00-00-F8-FC: Emit
Transaction timeout: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=5, NwkAddrOfInterest=44706, RequestType=0, StartIndex=0]": Emit
TX CMD: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=6, NwkAddrOfInterest=44706, RequestType=0, StartIndex=0]": Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0, cluster=1, addressMode=Device, radius=31, apsCounter=6, payload=6 162 174 0 0]": Emit
Slip Send:12-1B-00-1B-00-14-00-06-00-02-FD-FF-00-00-00-01-00-00-05-00-06-A2-AE-00-00-04-1F-21-FC: Emit
Slip Recv:12-1B-00-09-00-02-00-22-06-A0-FF: Emit
Slip Recv:0E-1C-00-07-00-A6-00-29-FF: Emit
Slip Send:04-1C-00-07-00-00-00-D9-FF: Emit
Slip Recv:04-1C-00-13-00-0C-00-22-06-02-FD-FF-00-00-A6-00-00-00-00-F5-FC: Emit
Transaction timeout: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=6, NwkAddrOfInterest=44706, RequestType=0, StartIndex=0]": Emit
TX CMD: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=7, NwkAddrOfInterest=44706, RequestType=0, StartIndex=0]": Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0, cluster=1, addressMode=Device, radius=31, apsCounter=7, payload=7 162 174 0 0]": Emit
Slip Send:12-1D-00-1B-00-14-00-07-00-02-FD-FF-00-00-00-01-00-00-05-00-07-A2-AE-00-00-04-1F-1D-FC: Emit
Slip Recv:12-1D-00-09-00-02-00-22-07-9D-FF: Emit
Slip Recv:0E-1E-00-07-00-A6-00-27-FF: Emit
Slip Send:04-1E-00-07-00-00-00-D7-FF: Emit
Slip Recv:04-1E-00-13-00-0C-00-22-07-02-FD-FF-00-00-A6-00-00-00-00-F2-FC: Emit
Transaction timeout: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=7, NwkAddrOfInterest=44706, RequestType=0, StartIndex=0]": Emit
TX CMD: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=8, NwkAddrOfInterest=44706, RequestType=0, StartIndex=0]": Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0, cluster=1, addressMode=Device, radius=31, apsCounter=8, payload=8 162 174 0 0]": Emit
Slip Send:12-1F-00-1B-00-14-00-08-00-02-FD-FF-00-00-00-01-00-00-05-00-08-A2-AE-00-00-04-1F-19-FC: Emit
[... cut ...]
RX APS: "ZigBeeApsFrame [sourceAddress=44706/1, destinationAddress=0/1, profile=260, cluster=1027, addressMode=0, radius=0, apsCounter=0, payload=24 56 10 0 0 41 245 3 20 0 40 255 16 0 41 150 39]": Emit
Incoming message from unknown node "AEA2": Notifying announce listeners: Emit
44706: NWK Discovery node discovery already in progress: Emit
RX ZCL: "ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=False, direction=SERVER_TO_CLIENT, disableDefaultResponse=True, manufacturerCode=0, sequenceNumber=56, commandId=10]": Emit
Unknown node 44706: Emit
Incoming message did not translate to command.: Emit
`

TheBlubb14, 2020-08-07:
One quick press on the sensor button will just send the current data to the connected network. You have to long press the sensor button for i think 3 seconds to start the pairing mode. Then you should be able to discover and connect it with the playground

jgmdavies, 2020-08-07:
Thanks @TheBlubb14.

So far nothing happens in Playground when I do the long press to pair.  But deCONZ is fine immediately after, the sensor just shows up.

jgmdavies, 2020-08-07:
If I use the 'join' command in Playground and then try to manually pair the sensor (long press), I get:

```
Slip Recv:1C-1A-00-12-00-0B-00-03-98-AF-85-04-00-8D-15-00-FF-E7-52-FB: Emit
Slip Recv:1C-1B-00-0C-00-05-00-02-25-C5-FF-E7-E6-FC: Emit
Slip Recv:0E-1C-00-07-00-AA-00-25-FF: Emit
Slip Send:17-1A-00-07-00-00-00-C8-FF: Emit
Slip Recv:17-1A-00-2A-00-23-00-22-02-FD-FF-00-02-25-C5-00-00-00-13-00-0C-00-26-25-C5-98-AF-85-04-00-8D-15-00-80-00-AF-FF-7E-68-00-00-E7-DA-F4: Emit
RX APS: "ZigBeeApsFrame [sourceAddress=50469/0, destinationAddress=65533/0, profile=0, cluster=19, addressMode=0, radius=0, apsCounter=0, payload=38 37 197 152 175 133 4 0 141 21 0 128]": Emit
Incoming message from unknown node "C525": Notifying announce listeners: Emit
50469: NWK Discovery scheduling node discovery: Emit
50469: NWK Discovery starting node discovery: Emit
TX CMD: "IeeeAddressRequest [0/0 -> 65533/0, cluster=1, transId=6, NwkAddrOfInterest=50469, RequestType=0, StartIndex=0]": Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65533/0, profile=0, cluster=1, addressMode=Device, radius=31, apsCounter=6, payload=6 37 197 0 0]": Emit
RX CMD: "DeviceAnnounce [50469/0 -> 65533/0, cluster=19, transId=38, NwkAddrOfInterest=50469, IeeeAddr=00158D000485AF98, Capability=128]": Emit
DISCOVERY Extension: Mesh related command received. Triggering mesh update.: Emit
"00158D000485AF98": Device announce received. NWK=50469: Emit
Slip Recv:0E-1B-00-07-00-AA-00-26-FF: Emit
DISCOVERY Extension: Start mesh update task with interval of 60 seconds: Emit
Slip Send:17-1B-00-07-00-00-00-C7-FF: Emit
"00158D000485AF98": NWK Discovery add node 50469: Emit
Slip Recv:17-1B-00-35-00-2E-00-22-02-00-00-01-02-25-C5-01-04-01-00-00-17-00-18-00-0A-05-00-42-0C-6C-75-6D-69-2E-77-65-61-74-68-65-72-01-00-20-05-00-AF-FF-2F-68-00-00-E7-A1-F5: Emit
DISCOVERY Extension: Starting mesh update: Emit
RX APS: "ZigBeeApsFrame [sourceAddress=50469/1, destinationAddress=0/1, profile=260, cluster=0, addressMode=0, radius=0, apsCounter=0, payload=24 0 10 5 0 66 12 108 117 109 105 46 119 101 97 116 104 101 114 1 0 32 5]": Emit
"00212EFFFF042B9F": DISCOVERY Extension: Starting mesh update: Emit
Incoming message from unknown node "C525": Notifying announce listeners: Emit
"00212EFFFF042B9F": Node SVC Discovery: Update mesh: Emit
"00158D000485AF98": Node state updated from UNKNOWN to ONLINE: Emit
50469: NWK Discovery node discovery already in progress: Emit
"00158D000485AF98": Updating Node 50469: Emit
"00212EFFFF042B9F": Node SVC Discovery: starting new tasks [NEIGHBORS, ROUTES]: Emit
"00212EFFFF042B9F": Node SVC Discovery: scheduled [NEIGHBORS, ROUTES]: Emit
"00158D000485AF98": Node sending added: Emit
"00158D000485AF98": Node sending added: Emit
Node 00158D000485AF98 added ZigBeeNode [state=ONLINE, IEEE=00158D000485AF98, NWK=0xC525]
"00158D000485AF98": Node sending added: Emit
"00212EFFFF042B9F": Node SVC Discovery: running: Emit
"00158D000485AF98": Data store: Deferring write for 250ms.: Emit
RX ZCL: "ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=False, direction=SERVER_TO_CLIENT, disableDefaultResponse=True, manufacturerCode=0, sequenceNumber=0, commandId=10]": Emit
TX CMD: "ManagementLqiRequest [0/0 -> 0/0, cluster=49, transId=7, StartIndex=0]": Emit
TX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=49, addressMode=Device, radius=31, apsCounter=7, payload=7 0]": Emit
"00158D000485AF98": DISCOVERY Extension: Adding discoverer for added node: Emit
"00158D000485AF98": Node SVC Discovery: start discovery: Emit
Unknown endpoint 1: Emit
"00158D000485AF98": Node SVC Discovery: starting new tasks [NWK_ADDRESS, NODE_DESCRIPTOR, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS]: Emit
Incoming message did not translate to command.: Emit
"00158D000485AF98": Node SVC Discovery: scheduled [NWK_ADDRESS, NODE_DESCRIPTOR, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS]: Emit
"00158D000485AF98": Node SVC Discovery: running: Emit
Slip Send:12-1C-00-1B-00-14-00-06-00-02-FD-FF-00-00-00-01-00-00-05-00-06-25-C5-00-00-04-1F-86-FC: Emit
Slip Recv:12-1C-00-09-00-02-00-22-06-9F-FF: Emit
Slip Recv:0E-1D-00-07-00-A6-00-28-FF: Emit
"00158D000485AF98": Node SVC Discovery: request NWK_ADDRESS failed. Retry 1, wait 1 ms before retry.: Emit
Slip Recv:12-1D-00-09-00-02-00-26-07-99-FF: Emit
Slip Send:12-1D-00-1E-00-17-00-07-00-03-9F-2B-04-FF-FF-2E-21-00-00-00-00-31-00-00-02-00-07-00-04-1F-1A-FC: Emit
Slip Recv:0E-1E-00-07-00-AE-00-1F-FF: Emit
"00158D000485AF98": Node SVC Discovery: running: Emit
Slip Recv:04-1E-00-13-00-0C-00-2E-06-02-FD-FF-00-00-A6-00-00-00-00-E7-FC: Emit
"00158D000485AF98": Node SVC Discovery: request NWK_ADDRESS failed. Retry 2, wait 1 ms before retry.: Emit
Slip Send:04-1E-00-07-00-00-00-D7-FF: Emit
"00158D000485AF98": Node SVC Discovery: running: Emit
Slip Send:17-1F-00-07-00-00-00-C3-FF: Emit
"00158D000485AF98": Node SVC Discovery: request NWK_ADDRESS failed. Retry 3, wait 2 ms before retry.: Emit
Slip Recv:17-1F-00-39-00-32-00-26-02-00-00-00-02-00-00-00-00-00-31-80-1B-00-07-00-01-00-01-9F-2B-04-FF-FF-2E-21-00-98-AF-85-04-00-8D-15-00-25-C5-12-00-01-FF-00-AF-FF-2F-68-00-00-E7-AB-F3: Emit
"00158D000485AF98": Node SVC Discovery: running: Emit
RX APS: "ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0, cluster=32817, addressMode=0, radius=0, apsCounter=0, payload=7 0 1 0 1 159 43 4 255 255 46 33 0 152 175 133 4 0 141 21 0 37 197 18 0 1 255]": Emit
[... cut ...]
50469: NWK Discovery ending node discovery: Emit
Slip Recv:1C-3C-00-0C-00-05-00-02-25-C5-FF-F6-B6-FC: Emit
Slip Recv:1C-3D-00-0C-00-05-00-02-25-C5-FF-F6-B5-FC: Emit
Slip Recv:1C-3E-00-0C-00-05-00-02-25-C5-FF-F6-B4-FC: Emit
Slip Recv:1C-3F-00-0C-00-05-00-02-25-C5-FF-F6-B3-FC: Emit
Slip Recv:1C-40-00-0C-00-05-00-02-25-C5-FF-F6-B2-FC: Emit
Slip Recv:1C-41-00-0C-00-05-00-02-25-C5-FF-F6-B1-FC: Emit
Slip Recv:1C-42-00-0C-00-05-00-02-25-C5-FF-F6-B0-FC: Emit
Slip Recv:1C-43-00-0C-00-05-00-02-25-C5-FF-F6-AF-FC: Emit

```

atifarkas, 2021-03-20:
Hi @dobrou 

Please can you share the code, how to use sensor? My problem is that the humidity, temp and pressure clusters are missing...

Regards,
Attila