Mr-Markus/ZigbeeNet #122: NetworkManager.Initialize() deadlock running from F# Interactive

reinux, 2020-09-11:
Calling `networkManager.Initialize()` in F# Interactive causes it to deadlock.

Here's an async stack trace:

[image]

I've been trying to reproduce this in C# Interactive so that it's easier to repro, but it's not seeing `ZigBeeNet.Transport.Serial` at all for whatever reason.

Mr-Markus, 2020-09-13:
I do not have experience with F#, but mabye @nicolaiw has an idea?

nicolaiw, 2020-09-14:
Maybe I will find the time to have a look at this in the next view weeks. Does it just happen in the interactive window or did you tried it in a simple console application as well? 

Regards

reinux, 2020-09-14:
Thanks @nicolaiw!

It works fine in a console application, to the extent that `Initialize()` will succeed and start waiting for connections, though I still haven't been able to get it to find any endpoints on my devices (which I mentioned [here](https://github.com/Mr-Markus/ZigbeeNet/issues/117#issuecomment-690952571)).

I just prefer to work in F# Interactive so I can iterate and tinker really quickly.

I thought it might have something to do with the fact that F# Interactive, when running in Visual Studio, hooks into the Windows message pump, but I turned that off using the `--gui-` flag, as well as tried running `dotnet fsi` in a console window, and it still behaves the same.

reinux, 2020-10-30:
Just an FYI, this works just fine with [.NET Interactive](https://github.com/dotnet/interactive), which is really cool. Exploratory tinkering with Zigbee is really convenient.

nicolaiw, 2020-11-06:
I am sorry. Did not find the time to dig into this. Is this still an Issue? I did not used F# for a while :D

reinux, 2020-11-06:
I've kind of given up on F# Interactive, but it works just fine in .NET Interactive.

reinux, 2020-11-06:
Closing since it looks like F# Interactive is going to be abandoned soon in favor of .NET Interactive anyway.

closed 2020-11-06 by reinux