Mr-Markus/ZigbeeNet #123: Can't discover any endpoints using ConBee II

reinux, 2020-10-05:
I'm not able to discover any endpoints using ConBee II, either in my own code or using the playground.

Tried with a Tradfri light, a Tradfri remote and a Sinope dimmer.

Thought this might be related to https://github.com/Mr-Markus/ZigbeeNet/issues/117, but commenting the `await RequestNetworkAddress()` and returning either false or true instead doesn't work for me.

I have no trouble getting it to work with deCONZ/Phoscon, so I think the hardware should be fine.

Thanks for any help!

Mr-Markus, 2020-10-08:
@DavidKarlas can you help?

jgmdavies, 2020-10-10:
@reinux  I get the same with my ConBee II - always a problem with discovery using ZigbeeNet, but the h/w is fine with deCONZ/Phoscon.  I'll go back to it in a day or two and try again.
Jim

DavidKarlas, 2020-10-11:
Sorry for slow response everyone... I opened this PR as workaround problem with NetworkAddressRequest... https://github.com/Mr-Markus/ZigbeeNet/pull/124

I'm not sure how it ever worked for me, I think I didn't use Discoverer and Playground at time I was writing driver... And didn't use it since...

reinux, 2020-10-12:
Ooh, that works! Thanks!

Mr-Markus, 2020-10-12:
I merged PR #124 into develop. I would also create a new nuget package version for this today

reinux, 2020-10-21:
Closing because it's resolved. Thanks @DavidKarlas and @Mr-Markus !

closed 2020-10-21 by reinux