Mr-Markus/ZigbeeNet #130: Reporting several working lights/dimmers

reinux, 2020-10-21:
All working with a ConBee II.

- Ikea Tradfri E26 spectrum, on/off, level and color temperature
- Sinope Smart Dimmer, on/off and level
- Philips Hue White Ambiance A19, on/off, level and color temperature
- Sengled Smart LED Multicolor A19 Bulb, Hub Required, on/off

The Sengled is not very responsive for some reason. It seems to respond just fine from Phoscon, so something's not quite right there.

I'm also trying to get switches and a sensor working, but I seem to be missing something, so I'll submit another issue regarding that.

Mr-Markus, 2020-10-21:
I would edit our wiki page with this info. Thanks.

I have not heard of Sinope and Sengled. Maybe country specific manufacturer? And what do you mean with "Hub requiered"? Are they not compatible with ConBee directly?

reinux, 2020-10-21:
"Hub Required" is just what's in the title on Amazon. They just call it that to distinguish it from their WiFi version. My bad, I should have been clearer about that.

[Sinope](https://www.amazon.ca/gp/product/B07BN188C1/) and [Sengled](https://www.amazon.ca/gp/product/B073ZBYXKQ/) are available in the US and Canada on Amazon.

Mr-Markus, 2020-10-21:
Ah ok I understand. I haven't seen this manufacturers in Europe, yet.
But I am happy to see that this project is used all over the world :-)

I edited wiki here: https://github.com/Mr-Markus/ZigbeeNet/wiki/Supported-devices

reinux, 2020-10-21:
Shoot, sorry, not sure how I missed it but the Sengled supports color and level as well.

Mr-Markus, 2020-11-04:
Thanks for reporting

closed 2020-11-04 by Mr-Markus