Mr-Markus/ZigbeeNet #136: Namespace inconsistent

fruzicka, 2020-12-27:
Please change the namespace to ZigBeeNet, so it's conform with your other libraries.

Mr-Markus, 2020-12-27:
Any examples? You also can send an PR

fruzicka, 2020-12-27:
There are libraries with Zig**b**eeNet.Hardware.ConBee, others are with Zig**B**eeNet. 
I have also noticed ZigBeeNet.Tran**(s is missing)**port.SerialPort.ZigBeeSerialPort
Maybe not everything is from you. I'm using the latest nuget packages.

Mr-Markus, 2020-12-27:
Ah ok I understand. You are right. I will check this.
I had this capitalize problems at the beginning and it's difficult to fix this on Windows, because you have to set some special git config that changes will be applied. Luckily I am developing on Linux now and here you are able to fix it because Linux is case sensetive by default. I Hope that Nuget Package names can be fixed, too

fruzicka, 2020-12-27:
It's not a big issue, just to be nice and clean ;)