Mr-Markus/ZigbeeNet #138: Digi XBee 3 USB support

oising, 2021-01-12:
I just bought a Digi Xbee 3 USB key and unfortunately it doesn't seem to work:

[image]

@andreasfedermann  -- I see you wrote the initial implementation. Is there anyway that I could help in order to fix this? Thank you!

spudwebb, 2021-01-12:
are you sure the 115200 baud rate is correct (you can pass it as a command line parameter) ?
I have a Digi XStick that works with the Playground app with a baud rate set to 9600

oising, 2021-01-12:
works from 1200bps to 961kbps according to https://media.digikey.com/pdf/Data%20Sheets/Digi%20International%20PDFs/Digi_XBee_3_USB_Adapter_7-17-19.pdf

oising, 2021-01-12:
tried with 9600 8n1

[image]

oising, 2021-01-12:
Flashed to latest zigbee 3 stack (which is emberznet 6.7.x) -- everything is fine in XCTU tool

[image]

oising, 2021-01-14:
So, I am reading the specifics of the AT command set for Xbee and I see that it's **not** in Command mode by default (it's in "Transparent" mode) -- you have to issue `+++` on the line and wait 1 second for the OK to return before sending AT commands - such as a software reset FR (the point at where it's failing right now) -- I can't see anywhere offhand where you might be sending the `+++` sequence -- is it possible that the Xbee model this was tested on is in Command mode by default?

oising, 2021-01-15:
Ah, so switching the XBee3 into API mode (default is Transparent mode) allowed the playground to initialize... I also had the chip in coordinator mode and not router but I'm not sure if that is an issue. API mode works. I set the mode using XCTU tool.

closed 2021-01-15 by oising