Mr-Markus/ZigbeeNet #143: NCP requires unsupported version of EZSP

Mallik-Tarigopula, 2021-02-21:
I am running NCP on MGM210PA32 Module from Silabs. I have connected NCP to my PC over UART. Below is the log when I run my .Net application. I got an error "NCP requires unsupported version of EZSP". I did not find any option in simplicity studio to change the version of NCP. Can you please guide me how to resolve this issue. Any changes on .Net code will resolve this issue?

Enter COM Port: COM9
[08:05:31 DBG] Network state is updated to INITIALISING
[08:05:31 DBG] EZSP Dongle: Initialize with protocol ASH2.
[08:05:31 DBG] Opening port COM9 at 57600 baud with FLOWCONTROL_OUT_XONOFF
[08:05:31 DBG] ASH: Connect
[08:05:31 DBG] AshFrameHandler parser task started
[08:05:31 DBG] ASH: Reconnect
[08:05:31 DBG] --> TX ASH frame: AshFrameRst []
[08:05:31 DBG] TX EZSP: EzspVersionRequest [desiredProtocolVersion=4]
[08:05:31 DBG] ASH: TX EZSP queue size: 1
[08:05:33 DBG] <-- RX ASH frame: AshFrameRstAck [version=2, resetCode=11, Reset: Software]
[08:05:34 DBG] ASH: Connected
[08:05:34 DBG] Ember: Link State change to True, initialised=False, networkStateUp=False
[08:05:34 DBG] Ember: Link State change to True ignored.
[08:05:34 DBG] --> TX ASH frame: AshFrameData [frmNum=0, ackNum=0, reTx=False, data=01 00 00 04]
[08:05:34 DBG] <-- RX ASH frame: AshFrameData [frmNum=0, ackNum=1, reTx=False, data=01 80 00 08 02 80 67]
[08:05:34 DBG] ASH: Frame acked and removed AshFrameData [frmNum=0, ackNum=0, reTx=False, data=01 00 00 04]
[08:05:34 DBG] EzspVersionResponse [protocolVersion=8, stackType=2, stackVersion=0x6780]
[08:05:34 DBG] RX EZSP: EzspVersionResponse [protocolVersion=8, stackType=2, stackVersion=0x6780]
[08:05:34 ERR] EZSP Dongle: NCP requires unsupported version of EZSP (required = V8, supported = V4)
[08:05:34 DBG] --> TX ASH frame: AshFrameAck [ackNum=1, notRdy=False]
[08:05:34 DBG] Network state is updated to OFFLINE
[08:05:34 ERR] Cannot add supported client cluster 6 when network state is OFFLINE
[08:05:34 ERR] Cannot add supported client cluster 768 when network state is OFFLINE
[08:05:34 ERR] Cannot add supported client cluster 1280 when network state is OFFLINE
PAN ID           = 0
Extended PAN ID  =
Channel          = CHANNEL_00
[08:05:34 DBG] TX EZSP: EzspGetKeyRequest [keyType=EMBER_CURRENT_NETWORK_KEY]
[08:05:34 DBG] ASH: TX EZSP queue size: 1
[08:05:34 DBG] --> TX ASH frame: AshFrameData [frmNum=1, ackNum=1, reTx=False, data=02 00 6A 03]
[08:05:34 DBG] <-- RX ASH frame: AshFrameData [frmNum=1, ackNum=2, reTx=False, data=02 80 01 58]
[08:05:34 DBG] ASH: Frame acked and removed AshFrameData [frmNum=1, ackNum=1, reTx=False, data=02 00 6A 03]
[08:05:34 DBG] RX EZSP: EzspGetLibraryStatusResponse [status=UNKNOWN]
[08:05:34 DBG] --> TX ASH frame: AshFrameAck [ackNum=2, notRdy=False]

quensen, 2024-02-08:
Hi @Mallik-Tarigopula,
did you solve this? Got the same issue with a _Sonoff Zigbee 3.0 USB Dongle Plus v2_.
Regards,
Stefan

quensen, 2024-09-14:
> any update? got the same issue :-(

I got rid of my issue. I guess by updating the dongle and also Zigbee2MQTT to the latest firmware version. But not sure, it‘s a long time ago.

What hardware do you use, @amosrohe ?

amosrohe, 2024-09-15:
> > any update? got the same issue :-(
> 
> I got rid of my issue. I guess by updating the dongle and also Zigbee2MQTT to the latest firmware version. But not sure, it‘s a long time ago.
> 
> What hardware do you use, @amosrohe ?

hi 
I am using the ZB-GW04 (v1.2)
 updating the firmware  will only change the "required" (to V13)  how did you change the supported ( supported = V4)
**I am referring to the "zigbeeNet" not  "Zigbee2MQTT"**

tnx Amos

MrZander, 2024-10-01:
I got past this error by manually setting the version:  `EzspFrame.SetEzspVersion(7);`
7 appears to be the max version this library supports, though.  I now get an IndexOutOfRangeException after it connects.

amosrohe, 2024-10-02:
hi 
I also tried to change the restriction in the code and during debugging, but it raised other issues (understandable since it is a different protocol version )

tibormohr, 2025-03-17:
Hi,

Any solutions? I have the same issue.