Mr-Markus/ZigbeeNet #146: Replace SeriLog with Microsoft.Extensions.Logging

oblaise, 2021-03-13:
Dear @Mr-Markus,

I'm opening this discussion to check if there is an interest in replacing `SeriLog` with the `Microsoft.Extensions.Logging` .NET standard 2.0 library. SeriLog is a very good library but that makes a requirement on a specific logging library which will not always be the adequate choice for some libraries usages. By using the `Microsoft.Extensions.Logging` abstraction the consumer of the library can use this mechanism of choice for logging (including SeriLog).

If you are interested I will be happy to implement this change on the `ZigBeeNet` and `ZigBeeNet.Hardware.TI.CC2531` libraries. That's the only one I can really validate myself for now because of my hardware.

In that case I would use the Singleton pattern approach to allow the logger to be called from everywhere in the code without requiring DI in all the classes that require logging. I think that implementing DI in the library will be far more complex because it will require propagation of the `ILogger[Factory]` interface instance to all classes that needs logging.

Please let me know what you thing about this proposal ?

nicolaiw, 2021-03-16:
Hi @Mr-Markus,

I had a look and it seems reasonable. There are much third party logging providers including, how @oblaise mentioned, serilog.

Third-party logging providers
Third-party logging frameworks that work with ASP.NET Core:

elmah.io (GitHub repo)
Gelf (GitHub repo)
JSNLog (GitHub repo)
KissLog.net (GitHub repo)
Log4Net (GitHub repo)
Loggr (GitHub repo)
NLog (GitHub repo)
PLogger (GitHub repo)
Sentry (GitHub repo)
Serilog (GitHub repo)
Stackdriver (Github repo)

source: https://docs.microsoft.com/de-de/aspnet/core/fundamentals/logging/?view=aspnetcore-5.0

@Mr-Markus what do you think?

Regards

Mr-Markus, 2021-03-17:
Yes, I think this would be a nice enhancment

Mr-Markus, 2021-04-11:
Done with #153

closed 2021-04-11 by Mr-Markus