Mr-Markus/ZigbeeNet #154: Reliable way to detect dongle disconnects

reinux, 2021-04-08:
It seems that the only way to detect when the dongle has disconnected is if a request returns null.

Is there a way we can detect when the dongle disconnects more reliably? Mine seems to be somewhat unstable, and occasionally requires restarting the NetworkManager.

jacobussystems, 2021-04-08:
@reinux  Is the instability maybe related to #152?

reinux, 2021-04-08:
I am using a ConBee II, so it may be, though it isn't as severe in my case -- only once every 10-20 hours as opposed to the 5-10 minutes that #152 reports.

Either way, I think it would be good to have a way to detect disconnects so that we can handle it properly. USB isn't always super reliable, after all.

What might be nice is an exception and a way to reset the connection synchronously.

jacobussystems, 2021-04-08:
Thanks - I agree re. disconnect detection.