Mr-Markus/ZigbeeNet #158: Get device state

decafbad86, 2021-07-06:
I am about to create an applicatiion with this library - most things work super easy but I do not know how to query the e.g. switched on / off state of a zigbee device. At the moment I am working with a iluminize 5120.1110 dimmer / switch module. The main thing at this module is the additional screw block where a wall switch can be attached to.

nicolaiw, 2021-07-09:
Hi @decafbad86 ,

the following snippet works for me with a CC2531 USB Stick and an OSRAM Plug.

```csharp
var onOfCluster = endpoint.GetInputCluster(ZclOnOffCluster.CLUSTER_ID);

if(onOfCluster != null)
{
    var state = await onOfCluster.ReadAttributeValue(ZclOnOffCluster.ATTR_ONOFF);
    Console.WriteLine($"state: {state}");
}
```

To test it you could insert the snippet above to the "read" command in the PlayGround app. Thats how I tested it.

see: https://github.com/Mr-Markus/ZigbeeNet/blob/develop/samples/ZigBeeNet.PlayGround/Program.cs#L517

Hint: If you are running the app e.g. from Visual Studio make sure to set your Zigbee-Dongle to the launchSettings.json

see: https://github.com/Mr-Markus/ZigbeeNet/blob/develop/samples/ZigBeeNet.PlayGround/Properties/launchSettings.json#L5

Hope this will help you :)

Regards

decafbad86, 2021-07-10:
Hi @nicolaiw, this works fine af! Just for info - ther returning type is bool at this call.

Thanks!

closed 2021-07-10 by decafbad86