Mr-Markus/ZigbeeNet #159: Nuget packages out of date?

JanneMattila, 2021-08-05:
I was testing ZigbeeNet using nuget packages with ConBee and couldn't make it work correctly. However, if I git clone and try using PlayGround it does work correctly. And I noticed bunch of changes in this area in last 6 months.

I noticed that there are 46 commits difference between `master` and `develop` branches. Would it be possible to merge and release newer versions of the packages? Or should I continue using direct source references for my app?

Thanks for the great library 👍!

Mr-Markus, 2021-08-08:
Hi @JanneMattila 
thanks for your interest in this library. 
Since there are some bigger changes within appvayor and .net core it does not build successfully and so there was no merge into master since that. I was thinking about switching to Github Actions, but I realized that I have to learn more about it to use it, which I haven't done yet.
So I recommend you to use the source code as reference for your project for the moment.
But you are welcome to help me to switch to Github Actions :-)

Best
Markus

JanneMattila, 2021-08-09:
> So I recommend you to use the source code as reference for your project for the moment.
Got it. I'll do that.

> But you are welcome to help me to switch to Github Actions :-)
Actually, I can try to help on this one. Let me study on your current `appveyor.yml` setup and try to provide alternative setup with GitHub Actions.

JanneMattila, 2021-08-10:
Created #160 to continue the discussion. We can close this.

closed 2021-08-10 by JanneMattila