Mr-Markus/ZigbeeNet #162: Make the classes less private & more protected so we can extend some classes.

genesiscz, 2021-09-03:
So I am in process of writing an app for Xbee-Pro S1 802.15.4, which probably has several differences between yours Xbee implementation and I need to be able to override the Initialize function from ZigBeeDongleXBee class. There are several properties such as _serialPort, _frameHandler which are not overridable/accessable which makes it really hard to extend the class & just rewrite to my own means without replicating/changing a lot of code. 

Is there a possibility you could make a lot of these properties public so we can feely extend them?

Mr-Markus, 2021-09-03:
Please describe a little bit more in detail which classes or fields do you need.
In general this library should be very flexible, so that extending will be possible. But because each user will need someting else not everything is there from beginning.

No, I haven't used ser2net, yet. For me it seems that it could be an own implementation of [IZigBeePort](https://github.com/Mr-Markus/ZigbeeNet/blob/develop/libraries/ZigBeeNet/Transport/IZigbeePort.cs)

See https://github.com/Mr-Markus/ZigbeeNet/blob/develop/libraries/ZigBeeNet.Transport.SerialPort/ZigBeeSerialPort.cs

genesiscz, 2021-09-03:
Yes, I did exactly that. Perfect. 

I stepped into a different problem: I need to override the ZigBeeDongleXbee's Initialize method, because I need to NOT send some packets (reset) & send more packets to setup it a little more. The _frameHandler could be public, so you can actually do some more low level stuff if you need to. 

I literally copied the whole class because if I just extend the class & try to implement the Initialize method, it won't call the child method but the parent one (because it has no virtual keyword).