Mr-Markus/ZigbeeNet #165: Where would you receive data from nodes?

StefanLundmark, 2021-09-09:
I'm trying to wrap my head around this. Let's say I have a motion sensor, where would the motion events be received?

Is the correct way to use a AttributeListener? I've added one to each input cluster but they're never getting called..

Mr-Markus, 2021-09-11:
Take a look at the wiki of this repository https://github.com/Mr-Markus/ZigbeeNet/wiki

StefanLundmark, 2021-09-11:
Sorry, I forgot to mention that I've already checked the Wiki and went through most of the public code.

The wiki has examples of querying, but not listening. I'm worried that wouldn't scale well enough for many devices but even then, the wiki only shows you how to fetch manufacturer name from a basic cluster. I see there are other clusters but how do I identiy what the other clusters are? How do I identify the attributes they hold? What value type? Maybe this is caused by a lack of understanding of Zigbee, but it would be nice for a little hand holding here, I Googled intensively for two days. :)

For instance, there are mentions of "bindings" and that it would be the solution I'm looking for, but there are no examples of this whatsoever.

Anyway. Thanks for creating and sharing the repo Markus!

Mr-Markus, 2021-09-11:
You can use the Playground app in this repository as an example. Within there you can find the CommandListener here:
https://github.com/Mr-Markus/ZigbeeNet/blob/1305b749bc9f289d5dc2247c51a8cd00cfed0936/samples/ZigBeeNet.PlayGround/Program.cs#L598-L604
Here you have to catch your relevant commands and handle it.

Which commands and clusters are available can be found in the official Zigbee ZCL documentation. You are also getting information about the device if it is paired, because then a discoverer will be started and returns which clusters are implemented by the device. This information is also stored in your choosen database, which is done by an interface for different type of database. The easiest for you would  be the JSON one [here](https://github.com/Mr-Markus/ZigbeeNet/blob/develop/libraries/ZigBeeNet.DataStore.Json/JsonNetworkDataStore.cs)

I would remind that this library is a framework you can build your application with and not a ready-to-use application.

StefanLundmark, 2021-09-11:
Hi Markus. I've used the Playground. Not sure if we're mixing up stuff here - a command is not the same thing as incoming data from a sensor, right? In any case, I've added that listener before and that method is never called using a Conbee II. The motion device is joined and I get clusters listed if I request them, but they contain nothing about what kind of attributes I can get. The database doesn't say either, it's just a bunch of values.

I understand this is a framework, I'm a software developer myself. My point is that I don't know how to consume the library because the comments don't give me enough information and a lot of the concepts in the library lack examples/unit tests. Just my few cents and thanks again!

Mr-Markus, 2021-09-11:
I think you don't understand Zigbee, but this will be necessary if you would use this library. Documentation is very well, but you need Zigbee knowledge. So I suggest you to read more about Zigbee and it's server client concept. Than you understand what a command is. 
Also the playground app was helpful for other developers yet. Beside that the wiki tells you all you need to know about this code (not Zigbee itself).

closed 2021-09-11 by Mr-Markus

StefanLundmark, 2021-09-11:
I don't need to understand the protocol to know that the documentation is lacking, sorry if this makes you upset.