Mr-Markus/ZigbeeNet #166: Correct settings for XBee?

reinux, 2021-09-10:
I'm using an XBee connected via USB, and I'm having a hard time getting it to connect. There are a lot of settings that I can configure in XCTU, some of which seem to be reflected in ZigBeeNet and some that aren't. Are there any specific settings to get it to work?

At the very least, I need to switch it to API Mode in order to get `Initialize()` to work at all. After that, I'm able to attach `DiscoveryExtension`, but `PermitJoin` doesn't seem to have any effect, even though it returns SUCCESS. I'm also unable to connect anything, whether it's a Tradfri or another XBee.

Mr-Markus, 2021-09-11:
I have no experience with XBee, because I only own a Texas Instrument CC2531 USB dongle. But with that I have the same problems, that sometimes startup and pairing works and sometimes not. For me there are some problems at manufactures side, because they don't implement ZigBee standard correctly at all.
Because of that it is more than difficult to handle all problems are reported here. I am sorry that I can't help you, but maybe anyone else could help. Because of that I added question and help wanted tags

reinux, 2021-09-13:
Nuts... Okay, I'll report back if I figure out what's going on.

I need to get something working even if it's just raw 802.15.4 for now, but we're hoping to have a working Zigbee stack eventually.